nasa/fprime #3101: 'fprime-util generate' fails with error "ModuleNotFoundError: No module named 'pkg_resources'"

ntrawny, 2025-01-02:
| | |
|:---|:---|
|**_F´ Version_**| v3.5.1 |
|**_Affected Component_**| fprime-util  |
---
## Problem Description

`fprime-util generate` fails on MacOS with error `ModuleNotFoundError: No module named 'pkg_resources'`

Error is fixed by manually executing
`pip install setuptools`

## Context / Environment

Execute `fprime-util version-check` and share the output.

```
(fprime-venv) ntrawny@MT-307826 MyProject % fprime-util version-check
Operating System: Darwin
CPU Architecture: x86_64
Platform: macOS-14.7.2-x86_64-i386-64bit-Mach-O
Python version: 3.13.1
CMake version: 3.31.3
Pip version: 24.3.1
Pip packages:
    fprime-tools==3.5.1
    fprime-gds==3.5.1
    fprime-fpp-*==2.2.1
Project submodules:
    https://github.com/nasa/fprime.git @ v3.5.1

```

## How to Reproduce

1. On MacOS, follow ['Hello World' Tutorial ](https://fprime.jpl.nasa.gov/latest/tutorials-hello-world/docs/hello-world/) until step "Building the New F´ Project"
2. Execute `fprime-util generate`

```
(fprime-venv) ntrawny@xxx MyProject % fprime-util generate
Traceback (most recent call last):
  File "~/fprime/MyProject/fprime-venv/bin/fprime-util", line 5, in <module>
    from fprime.util.__main__ import main
  File "~/fprime/MyProject/fprime-venv/lib/python3.13/site-packages/fprime/util/__main__.py", line 10, in <module>
    import fprime.util.cli
  File "~/fprime/MyProject/fprime-venv/lib/python3.13/site-packages/fprime/util/cli.py", line 20, in <module>
    from fprime.util.commands import (
    ...<5 lines>...
    )
  File "~fprime/MyProject/fprime-venv/lib/python3.13/site-packages/fprime/util/commands.py", line 20, in <module>
    import pkg_resources
ModuleNotFoundError: No module named 'pkg_resources'
```

## Expected Behavior

Successful command execution.

## Suggestions:

Error for my particular configuration is fixed by manually executing
`pip install setuptools`
and then (now successfully) rerunning
`fprime-util generate`

I understand from conversation with @timcanham that this behavior is python version dependent. 

Maybe there is a way that this can be fixed for all users, or the team could add a paragraph to [\#troubleshooting](https://fprime.jpl.nasa.gov/latest/getting-started/installing-fprime/#troubleshooting)? 

Thank you!

mprather, 2025-01-04:
I just ran into this problem on ubuntu. Solved it by adding `python3-setuptools` to the list of system packages.

mprather, 2025-01-06:
I need to correct my last comment. My environment was dirtied while trying to solve this issue. Adding a system package does *not* resolve the issue. It must be installed via pip into the virtual env.

Suggestion: since this seems like a relatively new problem (see [commit](https://github.com/nasa/fprime-tools/commit/b8911c38ae3f9d5a3b78254fde4ced624c1f8261#diff-ef46fa64dc10be2ff932b343a67e0ae3ecd026b192505b45a84ec50375bf920f)) and the tooling is dependent on the library, the easy fix is to add setuptools to the fprime/requirements.txt file. 

I tried this out in a clean docker environment and was able to verify the issue is properly resolved.

PR #3110 (2025-01-07): Fixing missing python packages. Fixes: #3101, #3102
| | |
|:---|:---|
|**_Related Issue(s)_**|  |
|**_Has Unit Tests (y/n)_**|  |
|**_Documentation Included (y/n)_**|  |

---
## Change Description

Adding `legacy-cgi` and `pkg_resources` to the `requirements.txt`. Fixes #3101 and #3102.
commit bf4b0515 (2025-01-07): Fixing missing python packages. Fixes: #3101, #3102 (#3110)

* Fixing missing python packages. Fixes: #3101, #3102

* Limiting legacy-cgi to python 3.13+

* Updating workflows for python 3.13

* Downgrading to last 3.8 setuptools package

* Reverting to 3.12 for macOs run
closed 2025-01-07 by LeStarch