nasa/fprime #3102: HelloWorld Tutorial: `fprime-util build -j4` fails with "ModuleNotFoundError: No module named 'cgi'"

ntrawny, 2025-01-02:
| | |
|:---|:---|
|**_F´ Version_**| v3.5.1 |
|**_Affected Component_**|  fprime-util |
---
## Problem Description

HelloWorld Tutorial: `fprime-util build -j4` fails with `ModuleNotFoundError: No module named 'cgi'`

cgi is no longer part of the Python standard library. It was [removed in Python 3.13](https://docs.python.org/3/whatsnew/3.13.html#whatsnew313-pep594) after being deprecated in Python 3.11. The removal was decided in [PEP 594](https://peps.python.org/pep-0594/).

Workaround is to manually
`pip install legacy-cgi`
This is a copy of the cgi module on PyPI, no longer maintained or supported by the core Python team.

## Context / Environment

Execute `fprime-util version-check` and share the output.

```
(fprime-venv) ntrawny@xxx MyProject % fprime-util version-check
Operating System: Darwin
CPU Architecture: x86_64
Platform: macOS-14.7.2-x86_64-i386-64bit-Mach-O
Python version: 3.13.1
CMake version: 3.31.3
Pip version: 24.3.1
Pip packages:
    fprime-tools==3.5.1
    fprime-gds==3.5.1
    fprime-fpp-*==2.2.1
Project submodules:
    https://github.com/nasa/fprime.git @ v3.5.1

```

## How to Reproduce

1. On MacOS, follow ['Hello World' Tutorial ](https://fprime.jpl.nasa.gov/latest/tutorials-hello-world/docs/hello-world/)until step "Adding The Hello World Component"
2. Execute `fprime-util build -j4`

```
(fprime-venv) ntrawny@xxx HelloWorldDeployment % fprime-util build -j4
make: *** No rule to make target `HelloWorldDeployment'.  Stop.
[CMAKE] CMake failed to detect target, attempting CMake cache refresh and retry
-- [python3] python3 found at: ~/fprime/MyProject/fprime-venv/bin/python3
-- [fpp-tools] fpp-depend found at: ~/fprime/MyProject/fprime-venv/bin/fpp-depend
-- [fpp-tools] fpp-to-xml found at: ~/fprime/MyProject/fprime-venv/bin/fpp-to-xml
-- [fpp-tools] fpp-to-cpp found at: ~/fprime/MyProject/fprime-venv/bin/fpp-to-cpp
-- [fpp-tools] fpp-to-dict found at: ~/fprime/MyProject/fprime-venv/bin/fpp-to-dict
-- [fpp-tools] fpp-locate-defs found at: ~/fprime/MyProject/fprime-venv/bin/fpp-locate-defs
-- [FPRIME] Module locations: ~/fprime/MyProject/fprime;~/fprime/MyProject
-- [FPRIME] Configuration module: ~/fprime/MyProject/fprime/config
-- [FPRIME] Installation directory: ~/fprime/MyProject/build-artifacts
-- Target build toolchain/platform: Darwin/Darwin
-- Including ~/fprime/MyProject/fprime/cmake/platform/Darwin.cmake
-- Requiring thread library
-- [sub-build] Performing sub-build: info-cache
-- [sub-build] Generating: info-cache with target/fpp_locs;target/fpp_depend
-- [sub-build] Executing: info-cache with fpp_locs
-- [sub-build] Executing: info-cache with fpp_depend
-- [sub-build] Performing sub-build: info-cache - DONE
-- [LIBRARY] Adding modules from F´ framework
-- Adding Library: config
-- Adding Library: Fw_Buffer
-- Adding Library: Fw_Cmd
-- Adding Library: Fw_Com
-- Adding Library: Fw_Dp
-- Adding Library: Fw_Log
-- Adding Library: Fw_Logger
[... 159 lines cut ...]
ModuleNotFoundError: No module named 'cgi'
ModuleNotFoundError: No module named 'cgi'
make[3]: *** [~/fprime/MyProject/fprime/Autocoders/Python/src/fprime_ac/generators/templates/arrays/array_cpp.py] Error 1
make[3]: *** Waiting for unfinished jobs....
make[3]: *** [~/fprime/MyProject/fprime/Autocoders/Python/src/fprime_ac/generators/templates/component/cpp.py] Error 1
make[3]: *** [~/fprime/MyProject/fprime/Autocoders/Python/src/fprime_ac/generators/templates/commands/CommandBody.py] Error 1
make[3]: *** [~/fprime/MyProject/fprime/Autocoders/Python/src/fprime_ac/generators/templates/channels/ChannelBody.py] Error 1
make[2]: *** [F-Prime/Autocoders/Python/src/fprime_ac/generators/templates/CMakeFiles/fprime_codegen.dir/all] Error 2
[  4%] Built target Fw_Cfg
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [HelloWorldDeployment/CMakeFiles/HelloWorldDeployment.dir/rule] Error 2
make: *** [HelloWorldDeployment] Error 2
[ERROR] CMake erred with return code 2
```

## Expected Behavior

Successful compilation of the HelloWorld Tutorial.

Manual workaround is:
`pip install legacy-cgi`
followed by (now successful)
`fprime-util build -j4`

More permanent fix would be to either update `requirements.txt` or replace cgi (maybe using suggestions from [here](https://docs.python.org/3/whatsnew/3.13.html#whatsnew313-pep594)?)

r-wells, 2025-01-04:
Thanks for finding + posting the workaround - saved me a lot of time

LeStarch, 2025-01-07:
@thomas-bc didn't we fix this with switching from `cheetah` to `C3`?

thomas-bc, 2025-01-07:
Fixed with the above PR