nasa/fprime #3116: JSON vs XML dict have different names for parameter cmds

zimri-leisher, 2025-01-07:
| | |
|:---|:---|
|**_F´ Version_**|3.5.0 |
|**_Affected Component_**| dictionaries  |
---
## Problem Description

In the XML dictionary, parameter set/save commands are named:
`component.XYZ_PRM_SET` and `component.XYZ_PRM_SAVE`

In the JSON dictionary, parameter set/save commands are named:
`component.XYZ_PARAM_SET` and `component.XYZ_PARAM_SAVE`

Is this an intentional change? It breaks all sequences that depend on the old names.

## Context / Environment

Execute `fprime-util version-check` and share the output.

```
Operating System: Linux
CPU Architecture: x86_64
Platform: Linux-5.15.133.1-microsoft-standard-WSL2-x86_64-with-glibc2.35
Python version: 3.10.12
CMake version: 3.22.1
Pip version: 22.0.2
Pip packages:
    fprime-tools==3.5.1
    fprime-gds==3.5.1
    fprime-fpp-*==2.2.1
Project submodules:```

## How to Reproduce

1. Build Ref topology
2. Observe difference between .xml and .json dicts

## Expected Behavior

They should be the same.

thomas-bc, 2025-01-08:
cc @jwest115 @bocchino

bocchino, 2025-01-08:
Looks like a discrepancy between the way FPP and the Python autocoder are auto-generating the parameter commands. I moved the issue here: https://github.com/nasa/fpp/issues/561.

closed 2025-01-08 by bocchino