nasa/fprime #3124: Module namespace mismatch between packets.xml and dictionary.json

zimri-leisher, 2025-01-09:
| | |
|:---|:---|
|**_F´ Version_**| 3.5.1 |
|**_Affected Component_**| autocoder/dictionary generator  |
---
## Problem Description

In a situation where:
1. Instances are declared outside of a module namespace
E.g.:
instances.fpp
```
instance bufMan: Svc.BufferManager <snip>
```
3. Instances are "used" inside of a module namespace
E.g.:
topology.fpp
```
module Ref {
instance bufMan
<snip>
}
```
The `packets.xml` will be required to reference telemetry channels via their fully-qualified name (e.g. `Ref.bufMan.tlmChan`), but the dictionaries generated will not include the module namespace (e.g. the dict will contain just `bufMan.tlmChan`.
This causes a mismatch, which prevents the GDS from running because it can't find channels in the dict that are in packets.xml.

## Context / Environment

Execute `fprime-util version-check` and share the output.

```
Operating System: Linux
CPU Architecture: x86_64
Platform: Linux-5.15.133.1-microsoft-standard-WSL2-x86_64-with-glibc2.35
Python version: 3.10.12
CMake version: 3.22.1
Pip version: 22.0.2
Pip packages:
    fprime-tools==3.5.1
    fprime-gds==3.5.1
    fprime-fpp-*==2.2.1
Project submodules:
```

## How to Reproduce

1. Modify Ref app to use TlmPacketizer
2. Remove the `module Ref` line from `instances.fpp`
3. Remove the Ref:: namespace and Ref_ prefix from relevant places in `RefTopology*pp` (just so that Ref compiles)
4. Compile Ref, and notice that the packets.xml contains the `Ref.` prefix, whereas the dictionaries (both JSON and XML) do not.
5. Run GDS:
```
$ fprime-gds --packet-spec Top/RefPackets.xml --dictionary build-artifacts/Linux/Ref/dict/RefTopologyDictionary.json  -n
[INFO] Ensuring comm[ip] Application is stable for at least 1 seconds
[INFO] Running Application: /home/zleisher/work/fprime/venv/bin/python3
[INFO] Ensuring HTML GUI is stable for at least 2 seconds
[INFO] Running Application: /home/zleisher/work/fprime/venv/bin/python3
[ERROR] Parsing error: Channel Ref.cmdDisp.CommandsDispatched in pkt CDH, but cannot be found in channel dictionary
[ERROR] Failed to run application: /home/zleisher/work/fprime/venv/bin/python3 -u -m flask run --host 127.0.0.1 --port 5000. Error: /home/zleisher/work/fprime/venv/bin/python3 stopped with code 1 sooner than 2 seconds.
[INFO] Shutting down F prime due to error. Failed to run HTML GUI
```

## Expected Behavior

I would expect the place where the instances are declared to correspond to their namespace, and for this behavior to be consistent across the dict and packets.xml

timcanham, 2025-01-09:
This is a band-aid until @bocchino finishes the FPP packets. The [latest](https://github.com/nasa/fprime/pull/3035) packet autocoder will strip the module from the channel name and look it up that way in the old XML dictionaries did, but leave it on for the GDS to parse.

zimri-leisher, 2025-01-09:
Ah gotcha. What release is that expected to be in? When it comes in, does that mean that we can remove the module specifiers from packets.xml channel names?

timcanham, 2025-01-09:
@bocchino can weigh in on the expected release. The new packet specifications will be in FPP, so all the namespaces will be consistent with FPP rules.

bocchino, 2025-01-09:
> What release is that expected to be in?

I'm working on the FPP packet specifications now. My current estimate is to have it released in FPP in April 2025. So it will go in the next F Prime release after that. So maybe within the next six months?

zimri-leisher, 2025-01-10:
Related:
What if I declare components and topologies outside of a module?
It seems to work, but this bandaid fix in tlm_packet_gen causes it to crash

[image]

timcanham, 2025-01-10:
Uh, so don’t do that for now? 😆

zimri-leisher, 2025-01-10:
Lol fair enough... was just trying to make it so that we didn't have to add "topologies." before every command in every sequence, and also in packets.xml. 

There's a conflict between the need to be linguistically precise (so use fully qualified names) and the need to be concise (using mnemonics). Maybe one day there could be a way to map names of FPrime objects (cmds, tlm, prms) into a global namespace meant for abbreviated names.

Like a file where you could say:

`alias Ref.componentY.commandX = Y.X`

LeStarch, 2025-06-11:
JSON is now baselined.

closed 2025-06-11 by LeStarch