nasa/fprime #3125: `fprime-bootstrap project` Should recognize existing repo before creating an empty one

mprather, 2025-01-10:
| | |
|:---|:---|
|**_F´ Version_**| 3.5.1|
|**_Affected Component_**|  |
---
## Feature Description
The project bootstrapper assumes that a new repo is needed. It should first check to see if one already exists and, if so, use it.

## Rationale
Many orgs have existing large repos. Further a lot of dev teams try to limit the use of submodules. It would be nice to incorporate an F´ project somewhere within an existing repo in a clean fashion. 

```
\- <largeRepo>
  |- product_library_1
  |- product_library_2
  |- tools_etc
  |- fprime_project_1

```

With the current implementation, the only way to "easily" bring in an F´ project is to setup a separate repo and bring in the project as a submodule. Right off the bat you are working 2 submodules for the one F´ project. Doable but may not be desired. 

The alternative is to kill the repo provisioned by the bootstrapper and then reestablish the submodule for the fprime repo itself. This brings the submodule count down to 1. This too is doable but also requires more work.

If the strapper can check for the existence of an repo, it would make integration with larger codebases easier.

thomas-bc, 2025-01-14:
Implemented by https://github.com/fprime-community/fprime-bootstrap/pull/15 🎉 

`fprime-bootstrap project --populate` now populates the current existing git repo with the F´ related content. Please let us know if you have any feedback.

closed 2025-01-14 by thomas-bc

mprather, 2025-01-15:
@thomas-bc Nice fix. 

Suggestion: this needs to be documented in the tutorial pages. For example, in step 2 of https://fprime.jpl.nasa.gov/latest/tutorials-hello-world/docs/hello-world/#1-creating-an-f-project a side note should be added.  If you would like me to write up the addition, let me know.

This should also tie into #3018 as well.