nasa/fprime #3128: Missing breaking changes in release notes

zimri-leisher, 2025-01-13:
| | |
|:---|:---|
|**_F´ Version_**| 3.5.x |
|**_Affected Component_**| Release notes  |
---
## Problem Description
It would be significantly easier to judge the difficulty of updating to a new release if the release notes had a comprehensive list of all breaking changes.
I suggest that release notes for either 3.5.1 or 3.5.0 should include the following notes:
* Fw::Logger::logMsg renamed to Fw::Logger::log
* Type of component `m_objName` from char* -> Fw::ObjectName
* Os::Log -> Os::Console
* Several Os::Queue methods renamed (and return types changed to more modern Fprime versions)
* Implicit constructor of Fw::StringBase made explicit (no longer can convert implicitly from const char* to some Fw::StringBase type)
* Unintentional change, but now all the PRM_SET/SAVE cmds are PARAM_SET/SAVE
* FPrime GDS version change should be mentioned, linking to changelog for FPrime GDS
   * Default arguments changed (now --zmq-transport is default)
   * Expects JSON dictionary by default instead of XML

Additionally, there is a mistake in the current release notes where stop and join get reversed in the "Task Changes" section

## Context / Environment

Execute `fprime-util version-check` and share the output.

```
Operating System: Linux
CPU Architecture: x86_64
Platform: Linux-5.15.133.1-microsoft-standard-WSL2-x86_64-with-glibc2.35
Python version: 3.10.12
CMake version: 3.22.1
Pip version: 22.0.2
Pip packages:
    fprime-tools==3.5.1
    fprime-gds==3.5.1
    fprime-fpp-*==2.2.1
Project submodules:```

LeStarch, 2025-01-15:
All updated! Thanks!

closed 2025-01-15 by LeStarch

zimri-leisher, 2025-01-18:
Awesome, thanks! Haha, didn't know there was an accessor for m_objName, we should have been using that and we wouldn't have experienced a breaking change.