nasa/fprime #3148: Parameter with string type appear to truncate at 40 chars

mprather, 2025-01-17:
| | |
|:---|:---|
|**_F´ Version_**| 3.5.1|
|**_Affected Component_**|   |
---
## Problem Description
Long string Parameters appear to be truncated at 40 characters.

fpp definition:
```
  param TransactionID: string size 100 default "some-value-more-than-forty-characters-long-for-use-with-my-app"
```

cpp usage:
```
  const auto id = host.paramGet_TransactionID(isValid);
```

## How to Reproduce

1. Define in fpp
1. In Component code, retrieve via the paramGet_TransactionID method

Result: value returned only has 40 chars.

## Expected Behavior

The original default string should be returned. This is the first time I've bumped into the 40+ char range with parameters. All others work as expected but this case is simply breaking. Is there something else I need to do in order to work with larger strings?

bocchino, 2025-01-18:
I think this is an expected behavior in the default F Prime configuration. The default max parameter string size is 40 characters. You can try increasing the max parameter string size here:

https://github.com/nasa/fprime/blob/334c52598461e1c1805d81ab453e22e81701bb41/config/FpConfig.h#L337-L340

mprather, 2025-01-18:
Thanks, Rob. I looked at the c++ reference and the fpp docs but there was no mention of the limit in the sections that discussed types or parameters.  It would be a good idea to add to the documentation todo list a task to document/reference these limits. 

Also thinking about this a bit more...
- the FPP parser notify the user they have exceeded the hard limit
- the parser should also catch the condition where the supplied default string exceeds the stated size (i.e. user sets size X but the default is >X)

These behaviors in addition to improved documentation would really help make it easier to a) learn and b) catch errors.

bocchino, 2025-01-18:
It looks like `FW_PARAM_STRING_MAX_SIZE` is documented here in Table 42 (though it's misspelled):

https://fprime.jpl.nasa.gov/latest/documentation/user-manual/framework/configuring-fprime/#fpconfigh

You are right that FPP could check more here. Right now it doesn't know anything about these config constants, which are defined in C++. It's on our to-do list to move these constants from C++ into FPP and add more checks.

mprather, 2025-01-18:
Yes, once you mentioned there was a relationship, it was easier to find the value (and others). I think it would be a good idea to put a footnote in the FPP and C++ docs stating that the params are limited by a constant. A cross-reference would have helped a lot today.

To close this out, for a quick test I added the appropriate definition and was able to get the larger string to persist. For demo purposes, I'm all set with a string >40 chars.

Thanks,

closed 2025-01-18 by mprather

mprather, 2025-01-18:
For general reference, here's the two locations that could use a note that states that parameter strings are limited in length by FW_PARAM_STRING_MAX_SIZE.

https://fprime.jpl.nasa.gov/latest/documentation/reference/api/cpp/html/class_fw_1_1_param_string.html
https://nasa.github.io/fpp/fpp-spec.html#Specifiers_Parameter-Specifiers