nasa/fprime #3149: Throttled event logging is not thread-safe

zimri-leisher, 2025-01-21:
| | |
|:---|:---|
|**_F´ Version_**| 3.4.3 |
|**_Affected Component_**|  All events with throttling |
---
## Problem Description

Adding `throttle X` to an event (in this example, ("YY_ZZ") generates the following code:
```
<snip>
  void ABCDEComponentBase ::
    log_XX_YY_ZZ()
  {
    // Check throttle value
    if (this->m_XX_YY_Throttle >= EVENTID_XX_YY_THROTTLE) {
      return;
    }
    else {
      this->m_XX_YY_Throttle++;
    }
<snip>
```
These are not thread-safe accesses to the `m_XX_YY_Throttle` variable, and so could result in dangerous behavior if two threads try to log the same event. Likely the best fix is to make `m_XX_YY_Throttle` atomic, and barring that, to add a lock around reads/writes to it.
## Context / Environment

Execute `fprime-util version-check` and share the output.

```
Operating System: Linux
CPU Architecture: x86_64
Platform: Linux-5.15.133.1-microsoft-standard-WSL2-x86_64-with-glibc2.35
Python version: 3.10.12
CMake version: 3.22.1
Pip version: 22.0.2
Pip packages:
    fprime-tools==3.4.4
    fprime-gds==3.4.3
    fprime-fpp-*==2.1.0a3```
```

zimri-leisher, 2025-01-21:
I believe this is also the case for telemetry writing when "update on change" is present.

LeStarch, 2025-02-03:
While this is clearly a bug, we need to study it to come up with the appropriate fix before approving for implementation.  Assigned myself for this task.

LeStarch, 2025-02-26:
I recommend using an atomic variable to support this counter and to use `fetch_add(1)` for the increment.

LeStarch, 2025-02-26:
For future reference, this can occur when a `sync` port invocation sends a throttled Event.

closed 2025-03-19 by bocchino