nasa/fprime #3156: fprime-util impl does not generate stubs for

timcanham, 2025-01-23:
| | |
|:---|:---|
|**_F´ Version_**| v3.5.0 |
|**_Affected Component_**| all  |
---
## Problem Description

When adding a port to component FPP, the `fprime-util impl` command does not generate stubs for the handler.

## Context / Environment

Execute `fprime-util version-check` and share the output.

```
Operating System: Linux
CPU Architecture: x86_64
Platform: Linux-5.15.133.1-microsoft-standard-WSL2-x86_64-with-glibc2.35
Python version: 3.10.12
CMake version: 3.31.3
Pip version: 22.0.2
Pip packages:
    fprime-tools==3.5.0
    fprime-gds==3.5.0
    fprime-fpp-*==2.2.0
```

## How to Reproduce

1. Put an internal port in a component's FPP
2. Run `fprime-util impl`
3. Handler stubs are missing from template files

## Expected Behavior

Template files would contain stubs.

thomas-bc, 2025-01-23:
@bocchino

bocchino, 2025-01-23:
I am not able to reproduce the behavior described. When I add this to `Svc/CmdDispatcher/CmdDispatcher.fpp`
```
    internal port P (x: U32)
```
and run `fprime-util impl` in `Svc/CmdDispatcher`, I see this in `CommandDispatcher.template.hpp`
```
    // ----------------------------------------------------------------------
    // Handler implementations for user-defined internal interfaces
    // ----------------------------------------------------------------------

    //! Handler implementation for P
    void P_internalInterfaceHandler(U32 x) override;
```
and this in `CommandDispatcher.template.cpp`:
```
    // ----------------------------------------------------------------------
    // Handler implementations for user-defined internal interfaces
    // ----------------------------------------------------------------------

    //! Handler implementation for P
    void P_internalInterfaceHandler(U32 x) override;
```

timcanham, 2025-01-28:
Nevermind... I ran it again and found the function. Probably pilot error.

closed 2025-01-28 by timcanham