nasa/fprime #3159: Added to many telemetry channels?

Gonta01, 2025-01-24:
I added a lot of telemetry channels to my component and as a result the following assert fails inside TlmChan.cpp:

[image]

Could somebody give recommendations on how properly resolve this issue? Can I increase possible amount of telemetry or I should just accept limitations?

zimri-leisher, 2025-01-24:
Yep! It's easy to increase this number. All you have to do is modify the `fprime_config/TlmChanImplCfg.hpp` file:

[image]

Change the `TLMCHAN_HASH_BUCKETS` constant to a larger number. The only penalty is that your application takes up more memory.

zimri-leisher, 2025-01-24:
Also @LeStarch recommend moving this to discussion

Gonta01, 2025-01-27:
@zimri-leisher thank you for your answer.
How could issue be moved to discussion? I haven't found how to do it so far

closed 2025-01-27 by Gonta01