nasa/fprime #3164: Case-specific identifier names?

zimri-leisher, 2025-01-28:
| | |
|:---|:---|
|**_F´ Version_**| 3.5.1 |
|**_Affected Component_**| fprime identifier specification  |
---
## Problem Description

In an admittedly awful situation, if I wanted to name one parameter `param1` and another `Param1`, the FPrime specification does not mention whether this is valid, as it does not mention whether capitalization has syntactic meaning for identifiers. Right now, it seems like this would work, if it were not for the fact that the auto-generated PRM_SET/SAVE commands automatically capitalize the parameter names. This means that two parameters with a name which differs only by capitalization conflict because their SET/SAVE cmds have the same name.

Likewise, similar thing happens for telemetry. It is valid FPP, but the compilation step crashes because there are two enum consts named `CHANNELID_XXXX`, where `XXXX` is the name of the telemetry channel being duplicated.

## Context / Environment

Execute `fprime-util version-check` and share the output.

```
Operating System: Linux
CPU Architecture: x86_64
Platform: Linux-5.15.133.1-microsoft-standard-WSL2-x86_64-with-glibc2.35
Python version: 3.10.12
CMake version: 3.22.1
Pip version: 24.2
Pip packages:
    fprime-tools==3.5.1
    fprime-gds==0.1.dev1222+gb4c4122.d20250128
    fprime-fpp-*==2.3.0a1
Project submodules:```

## How to Reproduce

1. Go to Ref
2. Create a copy of a channel or param with a name which differs only by case
3. Observe error messages

## Expected Behavior

Maybe this should be caught at the FPP level?
This is certainly a minor problem, but it is worth mentioning.

LeStarch, 2025-01-28:
Agreed.  Thanks for posting this.  For now the recommendation would be to use case-insensitive uniqueness in name.  However, it is worth taking a look.

bocchino, 2025-01-28:
Note also that the current approach (allow `FooBar` and transform to `FOOBAR`) was inherited from the XML autocoders, which we are phasing out.

zimri-leisher, 2025-01-29:
I was thinking that this could be fixed by checking FPP identifier uniqueness case-insensitively, and adding such a qualifier to the FPP Language Spec/User's guide. That way, the behavior which is already implicitly not allowed becomes explicitly not allowed, and we don't have to worry about supporting it. I think in general I'm working under the assumption that all valid FPP code should have a valid CPP equivalent, but this is probably naive. I understand that you don't want design decisions about FPP to be made with reference to the CPP autocode, that makes sense to me.

LeStarch, 2025-02-03:
Current recommendation:
- Names should not be transformed by FPP at all. Users desiring UPPER_CASE_NAMES should use UPPER_CASE_PARAMETER_NAMES

closed 2025-04-01 by LeStarch