nasa/fprime #3174: GDS does not process 56 bits (or more) arguments properly

Shivaly-Reddy, 2025-01-30:
I have a command with an argument that is a struct. It contains U64/U8/bool. In fprime-gds when I tried to enter a U64 with  56 bits or more, it processes the argument incorrectly. However, it's fine with 52 bits and below. For example, 

U64 --> 0xffffc002001004 gets converted to  ” 0xffffc00200100**0**" (56 bits)
U64 --> 0xfffc002001004 gets converted to 0xfffc002001004 (52 bits -- as expected)

LeStarch, 2025-01-30:
Moved to F Prim as it is a GDS issue.

LeStarch, 2025-01-30:
This is an issue with raw values in JavaScript, which are limited to 56 bits of precision.  We fixed the processing with scalars, but it appears we need to fix it with Arrays and Structs too.

arizvi786, 2025-01-31:
@LeStarch do we know when this issue can be made available in a release tag?

LeStarch, 2025-02-01:
We are still investigating a root cause for this issue.  Can probably schedule a path forward next week!

LeStarch, 2025-02-06:
@arizvi786 a PR is in to fix this issue, and will be included in the next release!

closed 2025-02-12 by LeStarch