nasa/fprime #3182: Memory leak is not on by default

kevin-f-ortega, 2025-02-05:
| | |
|:---|:---|
|**_F´ Version_**| 9a4b17b001d18727d34da3fcf5486a82c49609ef|
|**_Affected Component_**|   cmake |
---
## Problem Description

When I run my unit tests, it does not run the memory sanitation checks by default. I need to manually enable the flags using 
```bash
fprime-util generate --ut -DENABLE_SANITIZER_ADDRESS=ON -DENABLE_SANITIZER_LEAK=ON -DENABLE_SANITIZER_UNDEFINED_BEHAVIOR=ON
```

## Context / Environment

```
Operating System: Linux
CPU Architecture: x86_64
Platform: Linux-4.18.0-553.16.1.el8_10.x86_64-x86_64-with-glibc2.28
Python version: 3.11.9
CMake version: 3.27.4
Pip version: 25.0
Pip packages:
    fprime-tools==3.5.1
    fprime-gds==3.5.2a1
    fprime-fpp-*==2.3.0a1
Project submodules:
```

## How to Reproduce

1. Introduce a memory leak into your component
2. `fprime-util generate --ut`
3. `fprime-util check`
4. See no memory leak has been detected

## Expected Behavior

I expect fprime-util generate --ut will enable the sanitation by default.

zimri-leisher, 2025-02-06:
This is interesting, at least anecdotally memory leaks are detected by default for me when I run `fprime-util generate --ut`. Perhaps this has to do with the GTest or G++ version I'm on?

thomas-bc, 2025-02-06:
Mine are on by default as well... at some point `fprime-util` would inject these parameter values regardless of their default values in CMake... I'll look into it

LeStarch, 2025-02-10:
The error is a relatively new fix.  We cleaned out the code in `fprime-util` that was heavy-handed and let CMake handle the defaults, but CMake wasn't implemented correctly. 

There is a fix already!

closed 2025-02-10 by LeStarch