nasa/fprime #3187: Support overwriting stubs in fprime-util impl

celskeggs, 2025-02-06:
| | |
|:---|:---|
|**_F´ Version_**| v3.5.1 |
|**_Affected Component_**| fprime-util |
---
## Feature Description

I propose adding support to `fprime-util impl` for overwriting the `ComponentName.cpp` and `ComponentName.hpp` files instead of creating the files as `ComponentName.template.cpp` and `ComponentName.template.hpp`.

This should be opt-in so that users don't inadvertently overwrite code they intend to keep. For example, gated behind `--overwrite`.

## Rationale

When I create a new component, I've gotten into the habit of customizing the `.fpp` file after using the cookiecutter and before generating the implementation.

```
# fprime-util new --component
[INFO] Cookiecutter source: using builtin
  [1/8] Component name (MyComponent): ComponentName
  [2/8] Component short description (Component for F Prime FSW framework.): 
  [3/8] Component namespace (NamespaceName): 
  [4/8] Select component kind
    1 - active
    2 - passive
    3 - queued
    Choose from [1/2/3] (1): 2
  [5/8] Enable Commands?
    1 - yes
    2 - no
    Choose from [1/2] (1): 2
  [6/8] Enable Telemetry?
    1 - yes
    2 - no
    Choose from [1/2] (1): 2
  [7/8] Enable Events?
    1 - yes
    2 - no
    Choose from [1/2] (1): 2
  [8/8] Enable Parameters?
    1 - yes
    2 - no
    Choose from [1/2] (1): 2
[INFO] Found CMake file at '[project]/Components/NamespaceName/CMakeLists.txt'
Add ComponentName to [project]/Components/NamespaceName/CMakeLists.txt at end of file? (yes/no) [yes]: 
Generate implementation files? (yes/no) [yes]: no
[INFO] Did not generate implementations for /workspaces/[project]/Components/NamespaceName/ComponentName. Please do so manually.
# [... manually edit ComponentName.fpp ...]
# fprime-util impl
[0/1] Re-running CMake...
[...]
ninja: no work to do.
Formatting [1/2] /workspaces/[project]/Components/NamespaceName/ComponentName/ComponentName.template.cpp
Formatting [2/2] /workspaces/[project]/Components/NamespaceName/ComponentName/ComponentName.template.hpp
```

But since `fprime-util impl` has different behavior than the implementation phase of `fprime-util new --component`, I now need to manually rename `ComponentName.template.cpp` to `ComponentName.cpp` and `ComponentName.template.hpp` to `ComponentName.hpp`.

This request is obviously very low priority... but it would sure be convenient!

thomas-bc, 2025-02-06:
If anyone would like to contribute for this ticket, this is likely the place to look at https://github.com/nasa/fprime-tools/blob/devel/src/fprime/fpp/impl.py

FPP automatically adds the `.template.` suffix ( _mid_ fix?) I believe, but `fprime-util` could rename the files if it wanted to...

Tanuj-rai, 2025-05-19:
Hey @thomas-bc, I'd love to work on this issue. Could you please assign it to me?

Tanuj-rai, 2025-05-20:
Thanks @thomas-bc for assigning it to me. I will open the PR regarding it as soon as possible.

thomas-bc, 2025-05-24:
@Tanuj-rai thanks a lot! This looks like what I expected. Can you please open the PR in the main repo https://github.com/nasa/fprime-tools ?

closed 2025-08-21 by thomas-bc