nasa/fprime #3192: Cannot build fprime-zephyr reference project

ReggieMarr, 2025-02-10:
I was trying to follow along with the zephyr reference project so as to add support for more boards and I ran into [this issue](https://github.com/fprime-community/fprime-zephyr/issues/3):

Alternatively when I adjust the cmakelists.txt like so 

```
restrict_platforms(Zephyr)

add_fprime_subdirectory("${CMAKE_CURRENT_LIST_DIR}/Top/")
set(SOURCE_FILES "${CMAKE_CURRENT_LIST_DIR}/Stub.cpp")
set(MOD_DEPS ${FPRIME_CURRENT_MODULE}/Top)
register_fprime_deployment()

target_sources(app PRIVATE
    "${CMAKE_CURRENT_LIST_DIR}/Main.cpp"
)

target_link_libraries(LedBlinker_Top PUBLIC zephyr_interface)

add_dependencies(LedBlinker zephyr)
target_link_libraries(LedBlinker PUBLIC zephyr_interface)
target_link_libraries(LedBlinker PUBLIC zephyr)
set_target_properties(LedBlinker PROPERTIES
    ENABLE_EXPORTS ON
)

# Use Zephyr's link group mechanism
target_link_libraries(app PUBLIC
    LedBlinker_Top
    LedBlinker
)

set_target_properties(Svc_FatalHandler PROPERTIES EXCLUDE_FROM_ALL TRUE)

``` 

and build manually I encounter this error:

```
FAILED: bin/zephyr/LedBlinker 
: && ccache /home/user/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/arm-zephyr-eabi-g++  -gdwarf-4 LedBlinker/CMakeFiles/LedBlinker.dir/Stub.cpp.obj LedBlinker/CMakeFiles/LedBlinker.dir/__/fprime/Os/Stub/DefaultFile.cpp.obj LedBlinker/CMakeFiles/LedBlinker.dir/__/fprime-baremetal/fprime-baremetal/Os/Baremetal/DefaultCpu.cpp.obj LedBlinker/CMakeFiles/LedBlinker.dir/__/fprime-baremetal/fprime-baremetal/Os/Baremetal/DefaultMemory.cpp.obj LedBlinker/CMakeFiles/LedBlinker.dir/__/fprime-zephyr/Zephyr/Os/DefaultQueue.cpp.obj LedBlinker/CMakeFiles/LedBlinker.dir/__/fprime-zephyr/Zephyr/Os/DefaultMutex.cpp.obj LedBlinker/CMakeFiles/LedBlinker.dir/__/fprime-zephyr/Zephyr/Os/DefaultTask.cpp.obj LedBlinker/CMakeFiles/LedBlinker.dir/__/fprime-zephyr/Zephyr/Os/DefaultConsole.cpp.obj LedBlinker/CMakeFiles/LedBlinker.dir/__/fprime-zephyr/Zephyr/Os/DefaultRawTime.cpp.obj -o bin/zephyr/LedBlinker  lib/zephyr/libconfig.a  lib/zephyr/libFw_Cfg.a  lib/zephyr/libFw_Types.a  lib/zephyr/libFw_Logger.a  lib/zephyr/libFw_Obj.a  lib/zephyr/libFw_Port.a  lib/zephyr/libFw_Time.a  lib/zephyr/libFw_Comp.a  lib/zephyr/libZephyr_Svc_ZephyrTime.a  lib/zephyr/libFw_Com.a  lib/zephyr/libFw_Tlm.a  lib/zephyr/libFw_Log.a  lib/zephyr/libFw_Cmd.a  lib/zephyr/libFw_Prm.a  lib/zephyr/libFw_Buffer.a  lib/zephyr/libDrv_Ports.a  lib/zephyr/libSvc_Sched.a  lib/zephyr/libComponents_Led.a  lib/zephyr/libSvc_Fatal.a  lib/zephyr/libfprime-baremetal_Svc_FatalHandler.a  lib/zephyr/libZephyr_Drv_ZephyrGpioDriver.a  lib/zephyr/libOs.a  lib/zephyr/libSvc_Cycle.a  lib/zephyr/libZephyr_Drv_ZephyrRateDriver.a  lib/zephyr/libDrv_ByteStreamDriverModel.a  lib/zephyr/libZephyr_Drv_ZephyrUartDriver.a  lib/zephyr/libSvc_Ping.a  lib/zephyr/libFw_CompQueued.a  lib/zephyr/libSvc_ActiveLogger.a  lib/zephyr/libSvc_AssertFatalAdapter.a  lib/zephyr/libSvc_CmdDispatcher.a  lib/zephyr/libUtils_Hash.a  lib/zephyr/libUtils_Types.a  lib/zephyr/libSvc_FramingProtocol.a  lib/zephyr/libSvc_Deframer.a  lib/zephyr/libFw_Ports_SuccessCondition.a  lib/zephyr/libSvc_Framer.a  lib/zephyr/libSvc_PassiveConsoleTextLogger.a  lib/zephyr/libSvc_PassiveRateGroup.a  lib/zephyr/libSvc_RateGroupDriver.a  lib/zephyr/libSvc_StaticMemory.a  lib/zephyr/libSvc_SystemResources.a  lib/zephyr/libSvc_TlmChan.a  lib/zephyr/libLedBlinker_Top.a  lib/zephyr/libOs_File_Stub.a  lib/zephyr/libOs_Cpu_Baremetal.a  lib/zephyr/libOs_Memory_Baremetal.a  lib/zephyr/libOs_Queue_Zephyr.a  lib/zephyr/libOs_Mutex_Zephyr.a  lib/zephyr/libOs_Task_Zephyr.a  lib/zephyr/libOs_Console_Zephyr.a  lib/zephyr/libOs_RawTime_Zephyr.a  zephyr/libzephyr.a  lib/zephyr/libZephyr_Svc_ZephyrTime.a  lib/zephyr/libComponents_Led.a  lib/zephyr/libfprime-baremetal_Svc_FatalHandler.a  lib/zephyr/libZephyr_Drv_ZephyrGpioDriver.a  lib/zephyr/libZephyr_Drv_ZephyrRateDriver.a  lib/zephyr/libDrv_Ports.a  lib/zephyr/libZephyr_Drv_ZephyrUartDriver.a  lib/zephyr/libSvc_ActiveLogger.a  lib/zephyr/libSvc_Fatal.a  lib/zephyr/libSvc_AssertFatalAdapter.a  lib/zephyr/libSvc_CmdDispatcher.a  lib/zephyr/libSvc_Deframer.a  lib/zephyr/libSvc_Framer.a  lib/zephyr/libDrv_ByteStreamDriverModel.a  lib/zephyr/libSvc_FramingProtocol.a  lib/zephyr/libUtils_Types.a  lib/zephyr/libFw_Ports_SuccessCondition.a  lib/zephyr/libSvc_PassiveConsoleTextLogger.a  lib/zephyr/libSvc_PassiveRateGroup.a  lib/zephyr/libSvc_RateGroupDriver.a  lib/zephyr/libSvc_Cycle.a  lib/zephyr/libSvc_StaticMemory.a  lib/zephyr/libSvc_SystemResources.a  lib/zephyr/libSvc_TlmChan.a  lib/zephyr/libSvc_Sched.a  lib/zephyr/libSvc_Ping.a  lib/zephyr/libFw_CompQueued.a  lib/zephyr/libFw_Comp.a  lib/zephyr/libOs.a  lib/zephyr/libOs_File_Common.a  lib/zephyr/libUtils_Hash.a  lib/zephyr/libOs_Cpu_Common.a  lib/zephyr/libOs_Memory_Common.a  lib/zephyr/libOs_Queue_Common.a  lib/zephyr/libOs_Mutex_Common.a  lib/zephyr/libOs_Task_Common.a  lib/zephyr/libOs_Console_Common.a  lib/zephyr/libOs_RawTime_Common.a  lib/zephyr/libFw_Buffer.a  lib/zephyr/libFw_Prm.a  lib/zephyr/libFw_Cmd.a  lib/zephyr/libFw_Log.a  lib/zephyr/libFw_Tlm.a  lib/zephyr/libFw_Com.a  lib/zephyr/libFw_Time.a  lib/zephyr/libFw_Port.a  lib/zephyr/libFw_Obj.a  lib/zephyr/libFw_Logger.a  lib/zephyr/libFw_Types.a  lib/zephyr/libFw_Cfg.a  lib/zephyr/libconfig.a  zephyr/kernel/libkernel.a  lib/zephyr/libOs_Zephyr_Shared.a  -mcpu=cortex-m7  -mthumb  -mabi=aapcs  -mfpu=fpv5-d16  -mfloat-abi=hard  -mfp16-format=ieee  -fuse-ld=bfd  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--no-relax  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn  -Wl,-no-pie  -L"/home/user/zephyr-sdk-0.17.0/arm-zephyr-eabi/arm-zephyr-eabi"/lib/thumb/v7e-m+dp/hard && cd /fprime-zephyr-reference/build/LedBlinker && /usr/bin/cmake -DCMAKE_INSTALL_COMPONENT=LedBlinker -P /fprime-zephyr-reference/build/cmake_install.cmake
/home/user/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: warning: cannot find entry symbol _start; defaulting to 0000000000008000
/home/user/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: lib/zephyr/libOs_Task_Common.a(Task.cpp.obj): in function `_GLOBAL__sub_I__ZN2Os13TaskInterface9ArgumentsC2ERKN2Fw10StringBaseEPFvPvES6_jjjj':
/fprime-zephyr-reference/fprime/Os/Task.cpp:54: undefined reference to `Os::Mutex::Mutex()'
/home/user/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: lib/zephyr/libOs_Task_Common.a(Task.cpp.obj): in function `__static_initialization_and_destruction_0':
/fprime-zephyr-reference/fprime/Os/Task.cpp:54: undefined reference to `__dso_handle'
/home/user/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: /fprime-zephyr-reference/fprime/Os/Task.cpp:54: undefined reference to `Os::Mutex::~Mutex()'
/home/user/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: /fprime-zephyr-reference/fprime/Os/Task.cpp:54: undefined reference to `__aeabi_atexit'
/home/user/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: bin/zephyr/LedBlinker: hidden symbol `__dso_handle' isn't defined
/home/user/zephyr-sdk-0.17.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: final link failed: bad value

```

It seems as though the executable must be the zephyr executable and then use the fprime build as a library. 

I looked a bit into the "How-To: Develop an F´ Library" but it doesn't really seem applicable for this case where we want to expose the deployment as a library.

If there's any other suggestions here I'd appreciate it.

ReggieMarr, 2025-02-14:
So I managed to get things to build for my [zephyr reference project fork](https://github.com/ReggieMarr/fprime-zephyr-reference), one of the core hurdles was that zephyr and fprime's build systems both really want the final target (in fprime's case deployment) to be an executable.

I got around this by adding my own ["generate" function in the deployments CMakeLists.txt](https://github.com/ReggieMarr/fprime-zephyr-reference/commit/0a823ab212eef6cc385839fd3c15b6fcd0d8d8ea#diff-7075d72dd1cb423ee837219fc24465321275d5eb3a3ab1564644699e3e38e3a1) which looks like so:
```c++
restrict_platforms(Zephyr)

add_fprime_subdirectory("${CMAKE_CURRENT_LIST_DIR}/Top/")
set(SOURCE_FILES "${CMAKE_CURRENT_LIST_DIR}/Stub.cpp")
set(MOD_DEPS ${FPRIME_CURRENT_MODULE}/Top)

# This results in creating a new library target based on this module's sources
register_fprime_module()

include("${FPRIME_FRAMEWORK_PATH}/cmake/target/build.cmake")
function(generate_zephyr_deployment TARGET_TYPE TARGET_NAME SOURCE_FILES DEPENDENCIES)
    get_nearest_build_root(${CMAKE_CURRENT_LIST_DIR})
    set_property(GLOBAL PROPERTY MODULE_DETECTION TRUE)

    # The function this is based off of would call add_executable or add_library here
    # but zephyr handles this for us already

    # Modules properties for posterity
    set_target_properties("${TARGET_NAME}" PROPERTIES
        IS_DEPLOYMENT_LIB TRUE
        FP_TYPE "Deployment"
    )
    set_property(GLOBAL APPEND PROPERTY FPRIME_MODULES ${TARGET_NAME})

    setup_single_target("build" "${TARGET_NAME}" "${SOURCE_FILES}" "${DEPENDENCIES}")
endfunction(generate_zephyr_deployment)

# Links LedBlinker_Top to zephyr interface library
target_link_libraries(LedBlinker PUBLIC zephyr_interface)

target_sources(app PRIVATE
    "${CMAKE_CURRENT_LIST_DIR}/Main.cpp"
)

generate_zephyr_deployment("Deployment" "app" "${SOURCE_FILES}" "${MOD_DEPS}")

# Use Zephyr's link group mechanism
target_link_libraries(app PUBLIC
    LedBlinker
)

set_target_properties(Svc_FatalHandler PROPERTIES EXCLUDE_FROM_ALL TRUE)

```

 and then [modifying the "is_target_library" function in fprime/cmake/utilities](https://github.com/ReggieMarr/fprime-zephyr-reference/commit/0a823ab212eef6cc385839fd3c15b6fcd0d8d8ea#diff-7075d72dd1cb423ee837219fc24465321275d5eb3a3ab1564644699e3e38e3a1) like so:

```c++
function(is_target_library OUTPUT TEST_TARGET)
    set("${OUTPUT}" FALSE PARENT_SCOPE)
    if (TARGET "${TEST_TARGET}")
        # Check if marked as deployment
        get_target_property(IS_DEPLOYMENT "${TEST_TARGET}" IS_DEPLOYMENT_LIB)
        if(IS_DEPLOYMENT)
            set("${OUTPUT}" FALSE PARENT_SCOPE)
            return()
        endif()

        # Original library check
        get_target_property(TARGET_TYPE "${TEST_TARGET}" TYPE)
        ends_with(IS_LIBRARY "${TARGET_TYPE}" "_LIBRARY")
        set("${OUTPUT}" "${IS_LIBRARY}" PARENT_SCOPE)
    endif()
endfunction()

```

in combination with setting the ```FPRIME_SUB_BUILD_TARGETS``` to avoid the install and util targets (which would cause conflicts)

This is all key in ensuring that the ```setup_executable_implementations``` gets called to link the os implementations to the zephyr "app" which also is linked to the LedBlinker library. Originally this was the deployment target but I instead called ```register_fprime_module``` making it a static library which I could force to have the right flags by linking against the ```zephyr_interface``` (something I picked up from [stackoverflow](https://stackoverflow.com/questions/70747892/adding-external-library-to-zephyr))

A more elegant solution would be adding something to the API to support registering deployments as applications. I started at this but figured I could use some input from @LeStarch and/or @ethancheez before getting too far.

Also on a related note @ethancheez I ran into some startup faults mostly related to tuning stack sizes, due to the way the ```Zephyr/Os/Task.cpp``` has been setup we're kinda reliant on malloc'ing task stacks at run time and there seems to be a similar approach with the ```ZephyrRateDriver``` where instead of leveraging the singleton pattern and defining the timer at compile time we do so at runtime.

I get that where you land on dynamic allocation for runtime flexibility vs static allocation for more determinism/reliability is dependent on your con-ops but for me I think I'm gonna want to either refactor these modules to support static definitions or just provide separate implementations which could be leveraged with ```choose_fprime_implementation```. 
However I'd be curious to hear some context on why you went with a more dynamic approach in case I'm about to run into some fprime "gotcha".

LeStarch, 2025-07-07:
The project: fprime-zephyr-reference now builds!

closed 2025-07-07 by LeStarch