nasa/fprime #3194: FileUplink

zimri-leisher, 2025-02-10:
| | |
|:---|:---|
|**_F´ Version_**| 3.5.1 |
|**_Affected Component_**| FileUplink |
---
## Feature Description
FileUplink has no command to cancel the uplink. If the transmitted file is not finished, then FileUplink gets stuck in "waiting for data" mode and won't accept new files. There should be a command, and telemetry as to its current state.

thomas-bc, 2025-12-03:
It's possible this simply needs a new feature on the GDS, to send a `CANCEL` packet (already exists in Svc.FileUplink) through the Uplink tab

zimri-leisher, 2025-12-04:
I would still recommend adding a manual CANCEL cmd to FileUplink so that sequences/etc can cancel it

bocchino, 2025-12-04:
If the goal is to support fault protection, possibly we could add a no-uplink mode where FileUplink cancels any pending file uplink and ignores all future incoming packets until it is commanded out of that mode. If we just send a CANCEL command, then either (1) the ground is not sending DATA packets, in which case the CANCEL is essentially a no op (the component is quiescent anyway); or (2) to the ground is sending DATA packets, in which case the component would produce a slew of errors.

However, it seems to me that leaving FileUplink in its current state during fault protection seems fine, since the component won't do anything unless it receives a packet from the ground. Maybe there's a corner case where there's a fault during file uplink, and fault protection could intervene before the ground becomes aware and stops sending packets.

closed 2025-12-19 by thomas-bc