nasa/fprime #3201: Os::Queue Enable/Disable High Watermark

kevin-f-ortega, 2025-02-11:
| | |
|:---|:---|
|**_F´ Version_**| 3.5.1 |
|**_Affected Component_**|  Os::Queue |
---
## Feature Description

Be able to enable or disable High Watermark updates in Os::Queue.

## Rationale

The High Watermark updates are done on every Os::Queue::send. The member variable that keeps track of the high watermark requires a mutex to update. This is a cost operation to do on every queue send. So having a way to disable the high watermark updates would be nice for some projects that want to save that additional overhead.

budhe888, 2025-02-12:
I'd like to upvote the need to disable this. For running on an RTOS, where we offload work to tasks from an ISR, this will not work since we cannot lock a mutex from inside the ISR context. Without a way to disable this, ISRs are not really usable.

kevin-f-ortega, 2025-02-18:
Decision:

`getHighWaterMark` will only be implemented on OSes that natively provide support for Queue's high watermark. This function will not be implemented on OS that do not support it like VxWorks.

closed 2025-02-18 by kevin-f-ortega