nasa/fprime #3213: InT Tool to generate deployment specific configs

kevin-f-ortega, 2025-02-12:
| | |
|:---|:---|
|**_F´ Version_**| 3.5.1 |
|**_Affected Component_**| InT |
---
## Feature Description

Need a Python tool that takes as input the `instances.fpp` file and creates the `test/int` directory within a deployment's directory and generate a Python configuration file in that directory. The configuration file will map components to their component instance name (this name is found in the instances.fpp).  

## Rationale

Component instance names can be different across different deployments. As we start developing InT scripts for each component, we'll need to implement them while being deployment agnostic. The scripts will utilize the configuration file to know the component's instance name. This is different than the current approach to hardcode the instance name in the InT script.

thomas-bc, 2025-05-14:
- [ ] update `get_deployment_name()` function to use the value in the Dictionary.json,  rather than using the directory name

kevin-f-ortega, 2025-05-15:
@ekswang @LeStarch I updated this ticket with a better specification of what we want. Let me know if you have any questions, concerns, or improvements.

ekswang, 2025-05-28:
Side note, future improvement or recommendation is to use topology `instances.fpp` for mapping.

closed 2025-07-22 by kevin-f-ortega