nasa/fprime #3230: ROSES - [Pattern]

LeStarch, 2025-02-13:
### Description

Student review of #3229

gorsheep, 2025-05-09:
The documentation briefly states that the Manager/Worker pattern ensures a component remains “highly responsive to the rest of the system.” It would be beneficial to expand on this rationale, particularly for readers who may be new to the framework or to real-time system design. For example, additional context around the implications of blocking behavior—such as the inability to respond to telemetry or command requests—could help clarify the value of responsiveness. It may also help to address a subtle point: while a component may remain responsive, it may still be occupied and unable to perform new work. Explaining this distinction more explicitly could strengthen the reader’s understanding of the pattern’s purpose.

gorsheep, 2025-05-09:
Toward the end of the documentation, three strategies are listed for handling multiple incoming requests: ignoring them, queuing them, or using a pool of workers. At present, only the first strategy is illustrated through an example. It would enhance the usefulness of this section to include examples or design sketches demonstrating how the other two strategies could be implemented. Even brief overviews or references to additional resources would be helpful for those considering more complex concurrency handling.

gorsheep, 2025-05-09:
The documentation could benefit from a clearer explanation of how commands and data are exchanged between the manager and the worker. For instance:

- Are commands accompanied by parameters such as file names or task-specific inputs?
- Is the worker’s behavior configurable, or is it expected to perform a fixed set of tasks?
- What conventions are recommended for structuring this interaction? Clarifying these design expectations would help readers implement the pattern more effectively and with fewer assumptions.

gorsheep, 2025-05-09:
While the documentation mentions categories like “file operations,” it might be helpful to include more detailed examples of scenarios in which this pattern is especially useful. For instance, describing a specific operation (e.g., compressing large log files, performing time-intensive data transformations, or analyzing images from onboard sensors) would make the use cases more tangible. Even brief, illustrative examples could make it easier for developers to relate the pattern to their own system needs.