nasa/fprime #3232: Health Checking Pattern Student Review

LeStarch, 2025-02-14:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  |
---
### Description

Student review of #3231

gorsheep, 2025-05-09:
The documentation states, _"An active component needing periodic health-checking … should be connected to the Svc.Health component."_ It would be helpful to clarify whether a project is expected to use a single instance of `Svc.Health`, or whether multiple instances are allowed or recommended. Additionally, please note explicitly whether `Svc.Health` must be manually added to the topology, or if it is included by default.

gorsheep, 2025-05-09:
The explanation of how timeout events are triggered could be made more explicit. It would benefit users to clearly state that two timeout thresholds are specified: the first triggers a `WARNING_HI` event, and the second triggers a `FATAL` event if the component still does not respond.

gorsheep, 2025-05-09:
There is inconsistency in how special F Prime keywords like `WARNING_HI` and `FATAL` are formatted—sometimes they are enclosed in code-style formatting (e.g., backticks), and other times not. It would enhance readability and consistency if these keywords were always highlighted using backticks or another standardized style.

gorsheep, 2025-05-09:
The documentation mentions that a `FATAL` event can lead to a system reset or other handling. However, it is unclear whether a reset is the default response. Please clarify that reset behavior (if desired) must be explicitly configured by the developer.

gorsheep, 2025-05-09:
The documentation recommends using the pattern for "critical active components" but does not elaborate on when or why it might be inappropriate. Since this pattern seems broadly useful, it would be helpful to include guidance or examples of situations where it may not be necessary or advisable to apply it.

closed 2025-09-03 by LeStarch