nasa/fprime #3235: Drv BlockDriver Doesnn' Switch Context from ISR to Component Thread

arizvi786, 2025-02-14:
| | |
|:---|:---|
|**_F´ Version_**| 3.6.0 and earlier versions|
|**_Affected Component_**|   Drv/BlockDriver|
---
## Problem Description

The Drv/BlockDriver component ends up calling the internalInterface port handler directly from static ISR function: https://github.com/nasa/fprime/blob/devel/Drv/BlockDriver/BlockDriverImpl.cpp#L44 instead of calling the internalInterfaceInvoke() function to allow for a context switch from ISR to component thread.

## Context / Environment

Execute `fprime-util version-check` and share the output.

```
(fprime) bash-5.2$ fprime-util version-check
Operating System: Linux
CPU Architecture: x86_64
Platform: Linux-6.8.0-51-generic-x86_64-with-glibc2.39
Python version: 3.12.3
CMake version: 3.26.4
Pip version: 24.0
Pip packages:
    fprime-tools==3.5.1
    fprime-gds==3.5.2a2
    fprime-fpp-*==2.3.0
Project submodules:
(fprime) bash-5.2$ 
```

## How to Reproduce

1. By code inspection of line linked in problem description section
2.
3.

## Expected Behavior

The internal port interface runs in ISR context with current implementation which seems unintended.

LeStarch, 2025-02-14:
To make sure I follow: this is an issue because the output ports (CycleOut, telemetry) could be hooked up to `async` or `sync` ports where the second case `sync` would be an issue. The only way to guarantee that we leave ISR context is to use the component's thread via an internal interface.

Thus, instead of calling the handler directly, we should formally queue a message using the internal interface and let the component's thread process the message.

Is this correct?

timcanham, 2025-02-14:
Correct - this is a bug that @arizvi786 discovered. It should call the invoke method instead:

```c++
      //! Internal interface base-class function for InterruptReport
      void InterruptReport_internalInterfaceInvoke(
          U32 interrupt //!< The interrupt register value
      );
```

This was meant to demonstrate a pattern whereby a driver gets an interrupt and sends it to a driver thread to react to it. This is the case where there is no immediate need to react. Other drivers that need to react in in the interrupt context can add an output port or other logic that can be added to the ISR call.

closed 2025-05-12 by LeStarch