nasa/fprime #3249: Topology visualizer improvements

thomas-bc, 2025-02-19:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  |
---
## Feature Description

- [ ] Visualize components by kind  (color, shape, something else?)
- [ ] Visualize ports by kind 
- [ ] Represent array of ports better

ReggieMarr, 2025-02-25:
If I can tag on something else to this I think a "diff" feature would be incredibly powerful (albeit potentially a can of worms).

It would be so nice to have a CI rule for MR's so that beyond seeing just what code in what files have changed but seeing where ports have been added/removed or where connections have changed in a manner aided by ```fprime-util visualize```

thomas-bc, 2025-08-11:
Closing - the https://github.com/fprime-community/vscode-fpp VSCode extension has these features built-in