nasa/fprime #3255: GDS - Ability to plot chart based on channel timestamp instead of wall clock

rlcheng, 2025-02-20:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  |
---
## Feature Description

Currently when selecting a channel plot, sclk of FSW needs to be set to wall clock or else nothing gets plotted.

It would be better if plots can work off of timestamps of channel telemetry so regardless if sclk is set or not. Or GDS can have a concept of ERT and plot against ground receive time.

## Rationale

There are situations where SCLK needs to be set to a particular time that is not current wall clock and still need the charting abilities of GDS. For example, recreating a test that has SCLK set to 2020 and doing a comparison of FSW performance using the charting capabilities of GDS.

LeStarch, 2025-02-26:
Plotting of non-realtime data is a common need.  I am inclined to say "yes" on this recommendation.  Perhaps with a dropdown for time:

1. Realtime
2. Based on first Sample time
3. Anchored to a supplied time

arizvi786, 2025-02-26:
@LeStarch it would be great to have this functionality and it's high priority for a task I'm working on.