nasa/fprime #3256: GDS - Ability to plot multiple values on the same chart

rlcheng, 2025-02-20:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  |
---
## Feature Description

Ability to plot multiple channel values on the same chart, with ability for user to choose line colors for each channel.

## Rationale

Having the ability to plot multiple values to a chart is helpful in displaying GN&C related items, e.g. position and velocity values in x, y, and z axis.

LeStarch, 2025-02-26:
We support automatic sync between charts.  The vendor software seems to support multiple lines per chart.  We need to decide if we want to support this or just direct users to the existing multi-chart approach.

arizvi786, 2025-02-26:
@LeStarch the ability to plot multiple channels on the same chart will definitely be handy for a task I'm working on.

thomas-bc, 2025-04-02:
This feature request has been disapproved by the CCB. The reason for that is that it is out of scope of the F Prime GDS which is a lightweight GDS, primarily intended for testing and getting started. This feature is also more complex to implement than it sounds; different scales, etc.

If you'd like to plot complex charts, you may either export the channel telemetry data and use a graphing library such as matplotlib, or use a full-blown GDS.

closed 2025-06-06 by thomas-bc