nasa/fprime #3258: tlm_packet_gen.py: python reports use of escaped # illegal

mwette, 2025-02-23:
| | |
|:---|:---|
|3.6.0| |
|Autocoders  |
---
## Problem Description

Got this from a `fprime-util build`:
```
/.../lib/fprime/Autocoders/Python/bin/tlm_packet_gen.py:60: SyntaxWarning: invalid escape sequence '\#'
```

## Context / Environment

Execute `fprime-util version-check` and share the output.

```
Operating System: Linux
CPU Architecture: x86_64
Platform: Linux-6.11.0-18-generic-x86_64-with-glibc2.40
Python version: 3.12.7
CMake version: 3.30.3
Pip version: 24.2
Pip packages:
    fprime-tools==3.6.0
    fprime-gds==3.6.0
    fprime-fpp-*==2.3.0
Project submodules:
```

## How to Reproduce

Try this:
```
#!/usr/bin/env python3

foo = """
#include <hello.h>
"""

bar = """
\#include <hello.h>
"""

if True:
    print(foo)
    print(bar)

```

## Expected Behavior

No errors.

mwette, 2025-02-23:
This seems to help:
```
diff --git a/Autocoders/Python/bin/tlm_packet_gen.py b/Autocoders/Python/bin/tlm_packet_gen.py
index 2502afe56..6427a19f5 100755
--- a/Autocoders/Python/bin/tlm_packet_gen.py
+++ b/Autocoders/Python/bin/tlm_packet_gen.py
@@ -39,10 +39,10 @@ from lxml import etree
 
 header_file_template = """
 
-\#ifndef ${packet_list_name}_header_h
-\#define ${packet_list_name}_header_h
+\\#ifndef ${packet_list_name}_header_h
+\\#define ${packet_list_name}_header_h
 
-\#include <Svc/TlmPacketizer/TlmPacketizerTypes.hpp>
+\\#include <Svc/TlmPacketizer/TlmPacketizerTypes.hpp>
 
 namespace ${packet_list_namespace} {
 
@@ -53,18 +53,18 @@ namespace ${packet_list_namespace} {
 
 }
 
-\#endif // ${packet_list_name}_header_h
+\\#endif // ${packet_list_name}_header_h
 
 """
 
 impl_file_template = """
 
-\#include <Svc/TlmPacketizer/TlmPacketizerTypes.hpp>
-\#include <${output_header}>
+\\#include <Svc/TlmPacketizer/TlmPacketizerTypes.hpp>
+\\#include <${output_header}>
 
-\#include <FpConfig.hpp>
-\#include <FpConfig.hpp>
-\#include <Fw/Time/Time.hpp>
+\\#include <FpConfig.hpp>
+\\#include <FpConfig.hpp>
+\\#include <Fw/Time/Time.hpp>
 
 // Verify packets not too large for ComBuffer
 // if this macro gives a compile error, that means the packets are too large
```

LeStarch, 2025-03-04:
Also use a later version of Python. (3.13)

thomas-bc, 2025-06-11:
Closing as `Autocoders/` has been removed