nasa/fprime #3267: Queue message out of order on some systems

kevin-f-ortega, 2025-02-25:
| | |
|:---|:---|
|**_F´ Version_**| 3.6.0|
|**_Affected Component_**| ActiveLogger  |
---
## Problem Description

Events have a priority of 1. This causes events to come out out of order on some systems that don't truly implement a priority queue (e.g., VxWorks)

## Context / Environment

Execute `fprime-util version-check` and share the output.

```
--INSERT HERE--
```

## How to Reproduce

1. Build fprime-vxworks-reference
2. Run on BBB
3. Send No-Op and see events are in reversed order

## Expected Behavior

Events are in correct order.

timcanham, 2025-03-12:
3 Options:

1) Hand-written queue
2) Require VxWorks to use Posix Queues
3) Reduce the guarantees in framework
4) Live with it with caveats

LeStarch, 2025-03-17:
Ran into an issue where external collaborator wanted to use priority as expected.

timcanham, 2025-04-07:
Here is my recommendation:

1. Have each component declare in the SDD explicitly if priorities are needed.
2. Have POSIX queues independently selectable from other POSIX features (like threads/mutexes) 2
3. Make sure the VxWorks Posix message queues are functional and separable like 2. VxWorks native queues should assert if a non-zero priority is specified.
4. Based on a project's usage of components from 1, they can chose to use VxWorks or Posix queues.

thomas-bc, 2025-04-11:
All is implemented from the recommendations except item 1) which is an ongoing task.

VxWorks queues allow binary priority (0|1) for the purposes of handling high priority interrupts; but assert on other priority values.

thomas-bc, 2025-04-11:
Work to be done: update style guidelines to reflect item 1)

closed 2025-06-05 by vincewoo