nasa/fprime #3293: Introduce parameter versioning

timcanham, 2025-02-28:
| | |
|:---|:---|
|**_F´ Version_**|v3.6 |
|**_Affected Component_**| n/a |
---
## Feature Description

Embed dictionary version in parameter files.

## Rationale

Parameters stored in `PrmDb` have no way of verifying they match the current dictionary. If a new software release alters existing parameter IDs or the specification for the parameter changes there is no way for the component or `PrmDb` to detect it. This change would add a dictionary tag to parameters so version mismatches can be detected.

zimri-leisher, 2025-03-03:
This is a really good point. Perhaps now the only way to do this is to save a FSW version parameter and compare it when params are loaded.

timcanham, 2025-03-03:
Ideally, it would be a dictionary version rather than a FSW version. There can be many incremental software builds with a stable dictionary, so making it every git commit or something like that could be very burdensome.