nasa/fprime #3324: OSAL Timers and Timer Manager Component

LeStarch, 2025-03-04:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  |
---
## Feature Description

We should add a timer OSAL interface and a TimerManager component that allows another component to:

1. Register a timer
2. Receive a callback from the TimerManager when it expires.

kevin-f-ortega, 2025-03-04:
I had the same thought.

kevin-f-ortega, 2025-03-26:
The OSAL Timer will have this public interface

```c++
//! \brief Start timer and set it to expire once
//!
//! \param interval: The interval in which the timer should expire
//! \param callback: The callback function to invoke once the timer expires
//! \param arg: The argument to provide to the callback function
//! \return: The status of the operation
Status once(Fw::TimeInterval interval, void* callback, void* arg);

//! \brief Start timer and set it to expire periodically
//!
//! \param interval: The interval in which the timer should expire
//! \param callback: The callback function to invoke once the timer expires
//! \param arg: The argument to provide to the callback function
//! \return: The status of the operation
Status repeat(Fw::TimeInterval interval, void* callback, void* arg);

//! \brief Stop the timer
//!
//! \return: The status of the operation
Status stop();
```