nasa/fprime #3326: Code scanning rule of `Use of basic integral type` does not work properly

thomas-bc, 2025-03-04:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  CI / CodeQL |
---
## Problem Description

See https://github.com/nasa/fprime/security/code-scanning/92711

`FwSizeType` is typedef'ed to `uint64_t` but the code scanning rule still catches it. 

It appears the rule definition is making the assumption [here](https://github.com/github/codeql/blob/795a2e11756f9246c3a9255b3e5b6bc862ba286b/cpp/ql/src/JPL_C/LOC-3/Rule%2017/BasicIntTypes.ql#L15) that the underlying types should be e.g. U64 instead of uint64_t.

We should:
- confirm the issue root cause (see description)
- decide how to move forward:
    - make a PR to the [`JPL_C` rules in CodeQL](https://github.com/github/codeql/blob/795a2e11756f9246c3a9255b3e5b6bc862ba286b/cpp/ql/src/JPL_C/LOC-3/Rule%2017/BasicIntTypes.ql#L15) ?
    - clone and own these rules, as they are written for C but we are adapting to C++ (e.g. the one-statement-per-line rule applies to C but not to C++) ?

vincewoo, 2025-04-15:
@thomas-bc It seems like I either don't have access to https://github.com/nasa/fprime/security/code-scanning/92711 could you grant me access?

thomas-bc, 2025-04-23:
Confirmed!

closed 2025-04-23 by thomas-bc