nasa/fprime #3331: Move fprime-vxworks-reference WdTimer to fprime-vxworks

kevin-f-ortega, 2025-03-04:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  |
---
## Feature Description

A quick description of the requested feature.

## Rationale

A rationale for why the feature should be implemented in F´.

LeStarch, 2025-03-13:
Can you do that on a single directory's history?  If not, no.

NhatNguyen, 2025-03-14:
I think  should be able to do it. I looked up examples of how to do it with `git filter-repo`.

NhatNguyen, 2025-03-14:
> Verify fprime-vxworks-reference still compiles and runs on the BeagleBone Black (BBB). Ask @kevin-f-ortega, @chuynh4duarte, or @LeStarch for help running on the BBB.

Does this mean I will not be able to generate and build this project with WSL?

These are the changes I've made so far against a fork of each repo:

- https://github.com/fprime-community/fprime-vxworks/compare/devel...NhatNguyen:fprime-vxworks:fprime-3331?expand=1
- https://github.com/fprime-community/fprime-vxworks-reference/compare/devel...NhatNguyen:fprime-vxworks-reference:fprime-3331?expand=1

LeStarch, 2025-03-26:
@kevin-f-ortega is this issue complete?

kevin-f-ortega, 2025-03-26:
This issue is now complete

closed 2025-04-02 by LeStarch