nasa/fprime #3360: clang-tidy version 18 detects errors in the fprime-devel code

bevinduckett, 2025-03-10:
| | |
|:---|:---|
|**_F´ Version_**|  fprime-devel branch on 3/10/2025 |
|**_Affected Component_**|   |
---
## Problem Description

When following the `static-analyzer` instructions on https://github.com/nasa/fprime/blob/devel/CONTRIBUTING.md, I could not install clang-tidy-12 via apt-get because that version was too old, so instead I installed clang-tidy version 18.1.8 as `clang-tidy` . Running `fprime-util generate -f --ut -DCMAKE_CXX_CLANG_TIDY=clang-tidy` and then ` fprime-util build --all --ut -j16` failed due to warnings found (the warnings are shown below) 
```
[ 14%] Building CXX object F-Prime/Svc/BufferRepeater/CMakeFiles/Svc_BufferRepeater.dir/BufferRepeater.cpp.o
/home/USER/dev/fprime/Svc/PassiveConsoleTextLogger/ConsoleTextLoggerImplCommon.cpp:15:21: error: Value stored to 'severityString' during its initialization is never read [clang-analyzer-deadcode.DeadStores,-warnings-as-errors]
   15 |         const char *severityString = "UNKNOWN";
      |                     ^~~~~~~~~~~~~~   ~~~~~~~~~
/home/USER/dev/fprime/Svc/PassiveConsoleTextLogger/ConsoleTextLoggerImplCommon.cpp:15:21: note: Value stored to 'severityString' during its initialization is never read
   15 |         const char *severityString = "UNKNOWN";
      |                     ^~~~~~~~~~~~~~   ~~~~~~~~~
72 warnings generated.
Suppressed 71 warnings (71 in non-user code).
Use -header-filter=.* to display errors from all non-system headers. Use -system-headers to display errors from system headers as well.
1 warning treated as error
gmake[2]: *** [F-Prime/Svc/PassiveConsoleTextLogger/CMakeFiles/Svc_PassiveConsoleTextLogger.dir/build.make:91: F-Prime/Svc/PassiveConsoleTextLogger/CMakeFiles/Svc_PassiveConsoleTextLogger.dir/ConsoleTextLoggerImplCommon.cpp.o] Error 1
[ 14%] Building CXX object F-Prime/Svc/BufferRepeater/CMakeFiles/Svc_BufferRepeater.dir/BufferRepeaterComponentAc.cpp.o
[ 14%] Building CXX object F-Prime/Drv/ByteStreamDriverModel/CMakeFiles/Drv_ByteStreamDriverModel.dir/ByteStreamReadyPortAc.cpp.o
[ 14%] Building CXX object F-Prime/Svc/FramingProtocol/CMakeFiles/Svc_FramingProtocol.dir/FprimeProtocol.cpp.o
[ 14%] Building CXX object F-Prime/Fw/FilePacket/CMakeFiles/Fw_FilePacket.dir/FilePacket.cpp.o
gmake[2]: *** Waiting for unfinished jobs....
gmake[1]: *** [CMakeFiles/Makefile2:10706: F-Prime/Svc/PassiveConsoleTextLogger/CMakeFiles/Svc_PassiveConsoleTextLogger.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
[ 14%] Building CXX object F-Prime/Fw/FilePacket/CMakeFiles/Fw_FilePacket.dir/Header.cpp.o
```

## Context / Environment

Execute `fprime-util version-check` and share the output.

```
~/dev/fprime$ fprime-util version-check
Operating System: Linux
CPU Architecture: x86_64
Platform: Linux-5.15.167.4-microsoft-standard-WSL2-x86_64-with-glibc2.39
Python version: 3.12.3
CMake version: 3.28.3
Pip version: 25.0.1
Pip packages:
    fprime-tools==3.6.1
    fprime-gds==3.6.1
    fprime-fpp-*==2.3.0
Project submodules:
```

```
~/dev/fprime$ clang-tidy --version
Ubuntu LLVM version 18.1.8
  Optimized build.
```

## How to Reproduce

1. Install clang-tidy version 18.1.8 or newer
2. Checkout the latest version of the devel branch & activate the fprime virtual env 
3. Execute  `fprime-util generate -f --ut -DCMAKE_CXX_CLANG_TIDY=clang-tidy` and then ` fprime-util build --all --ut -j16` 

## Expected Behavior

The expected behavior is for `fprime-util build --all --ut -j16` to run to completion with no errors.

zeroping, 2025-05-06:
This seems related enough to go here.

When using BusFSWPackets.xml, clang-tidy-18 doesn't like the autocoded deploymentname/Top/BusFSWPacketsAc.cpp:

```
deploymentname/Top/BusFSWPacketsAc.cpp:13:17: error: redundant void argument list in function definition [modernize-redundant-void-arg,-warnings-as-errors]
void check_func(void) {
                ^~~~

Seems easy to miss this one, since it only shows up with the telemetry packetizer.

thomas-bc, 2025-08-04:
Fixed (see above PRs)

closed 2025-08-04 by thomas-bc