nasa/fprime #3362: Add Ref example to F Prime website

timcanham, 2025-03-11:
| | |
|:---|:---|
|**_F´ Version_**| v3.6+ |
|**_Affected Component_**| n/a |
---
## Feature Description

Add directions on how to run Ref example on the F Prime website.

## Rationale

People may want to take F Prime for a quick spin without having to work through a demo.

thomas-bc, 2025-03-11:
Needs https://github.com/nasa/fprime/issues/3323 if we're going forward with this.

thomas-bc, 2025-03-31:
Searching for `Ref` in the search bar will yield the correct page, and we have implemented https://github.com/nasa/fprime/issues/3323 to fix the README.

Also, the direct URL is https://fprime.jpl.nasa.gov/devel/Ref/ or https://fprime.jpl.nasa.gov/latest/Ref

closed 2025-03-31 by thomas-bc