nasa/fprime #3364: Compilation errors in auto-generated Topology.cpp

bevinduckett, 2025-03-11:
| | |
|:---|:---|
|**_F´ Version_**| fprime-devel branch cloned 3/11/2025 |
|**_Affected Component_**|   |
---
## Problem Description

The "-pedantic" option has been added to the  options passed to c++ which causes the auto-generated <deployment>Topology.cpp (e.g. MyDeployment/Top/MyDeploymentTopology.cpp) file to fail to compile with the below warnings ( `fprime-util build -v` was used to provide additional info)

```
cd /home/USER/dev/fprime/build-fprime-automatic-native/MyDeployment/Top && /usr/bin/c++ -DFW_BAREMETAL_SCHEDULER=0 -DTGT_OS_TYPE_LINUX -I/home/USER/dev/fprime -I/home/USER/dev/fprime/config -I/home/USER/dev/fprime/build-fprime-automatic-native -I/home/USER/dev/fprime/build-fprime-automatic-native/F-Prime -I/home/USER/dev/fprime/build-fprime-automatic-native/config -isystem /home/USER/dev/fprime/cmake/platform/types -std=c++11 -Wall -Wextra -Werror -Wold-style-cast -Wno-unused-parameter -Wshadow -pedantic -Wconversion -Wno-format-truncation -DASSERT_FILE_ID=0xa9eade69 -DASSERT_RELATIVE_PATH='"MyDeployment/Top/MyDeploymentTopology.cpp"' -MD -MT MyDeployment/Top/CMakeFiles/MyDeployment_Top.dir/MyDeploymentTopology.cpp.o -MF CMakeFiles/MyDeployment_Top.dir/MyDeploymentTopology.cpp.o.d -o CMakeFiles/MyDeployment_Top.dir/MyDeploymentTopology.cpp.o -c /home/USER/dev/fprime/MyDeployment/Top/MyDeploymentTopology.cpp
/home/USER/dev/fprime/MyDeployment/Top/MyDeploymentTopology.cpp: In function ‘void configureTopology(const MyDeployment::TopologyState&)’:
/home/USER/dev/fprime/MyDeployment/Top/MyDeploymentTopology.cpp:125:38: error: C++ designated initializers only available with ‘-std=c++20’ or ‘-std=gnu++20’ [-Werror=c++20-extensions]
  125 |     configurationTable.entries[0] = {.depth = 100, .priority = 0};
      |                                      ^
/home/USER/dev/fprime/MyDeployment/Top/MyDeploymentTopology.cpp:125:52: error: C++ designated initializers only available with ‘-std=c++20’ or ‘-std=gnu++20’ [-Werror=c++20-extensions]
  125 |     configurationTable.entries[0] = {.depth = 100, .priority = 0};
      |                                                    ^
/home/USER/dev/fprime/MyDeployment/Top/MyDeploymentTopology.cpp:127:38: error: C++ designated initializers only available with ‘-std=c++20’ or ‘-std=gnu++20’ [-Werror=c++20-extensions]
  127 |     configurationTable.entries[1] = {.depth = 500, .priority = 2};
      |                                      ^
/home/USER/dev/fprime/MyDeployment/Top/MyDeploymentTopology.cpp:127:52: error: C++ designated initializers only available with ‘-std=c++20’ or ‘-std=gnu++20’ [-Werror=c++20-extensions]
  127 |     configurationTable.entries[1] = {.depth = 500, .priority = 2};
      |                                                    ^
/home/USER/dev/fprime/MyDeployment/Top/MyDeploymentTopology.cpp:129:38: error: C++ designated initializers only available with ‘-std=c++20’ or ‘-std=gnu++20’ [-Werror=c++20-extensions]
  129 |     configurationTable.entries[2] = {.depth = 100, .priority = 1};
      |                                      ^
/home/USER/dev/fprime/MyDeployment/Top/MyDeploymentTopology.cpp:129:52: error: C++ designated initializers only available with ‘-std=c++20’ or ‘-std=gnu++20’ [-Werror=c++20-extensions]
  129 |     configurationTable.entries[2] = {.depth = 100, .priority = 1};
```

## Context / Environment

Execute `fprime-util version-check` and share the output.

```
:~/dev/fprime$ fprime-util version-check
Operating System: Linux
CPU Architecture: x86_64
Platform: Linux-5.15.167.4-microsoft-standard-WSL2-x86_64-with-glibc2.39
Python version: 3.12.3
CMake version: 3.28.3
Pip version: 25.0.1
Pip packages:
    fprime-tools==3.6.1
    fprime-gds==3.6.1
    fprime-fpp-*==2.3.0
Project submodules:
```

## How to Reproduce

1. Checkout a clone of the devel branch from the fprime repo ("fprime-bootstrap clone --rename issue-repro https://github.com/nasa/fprime.git")
2. activate the venv (`. fprime-venv/bin/activate`)
3. generate & build the project (`fprime-util generate && fprime-util build`)
4. add a deployment (`fprime-util new --deployment`, hit <ENTER> to accept defaults for all prompts)
5. build the project (`fprime-util build -v`) FAILS
```
/home/USER/dev/issue-repro/MyDeployment/Top/MyDeploymentTopology.cpp: In function ‘void configureTopology(const MyDeployment::TopologyState&)’:
/home/USER/dev/issue-repro/MyDeployment/Top/MyDeploymentTopology.cpp:125:38: error: C++ designated initializers only available with ‘-std=c++20’ or ‘-std=gnu++20’ [-Werror=c++20-extensions]
  125 |     configurationTable.entries[0] = {.depth = 100, .priority = 0};
      |                                      ^
/home/USER/dev/issue-repro/MyDeployment/Top/MyDeploymentTopology.cpp:125:52: error: C++ designated initializers only available with ‘-std=c++20’ or ‘-std=gnu++20’ [-Werror=c++20-extensions]
  125 |     configurationTable.entries[0] = {.depth = 100, .priority = 0};
      |                                                    ^
/home/USER/dev/issue-repro/MyDeployment/Top/MyDeploymentTopology.cpp:127:38: error: C++ designated initializers only available with ‘-std=c++20’ or ‘-std=gnu++20’ [-Werror=c++20-extensions]
  127 |     configurationTable.entries[1] = {.depth = 500, .priority = 2};
      |                                      ^
/home/USER/dev/issue-repro/MyDeployment/Top/MyDeploymentTopology.cpp:127:52: error: C++ designated initializers only available with ‘-std=c++20’ or ‘-std=gnu++20’ [-Werror=c++20-extensions]
  127 |     configurationTable.entries[1] = {.depth = 500, .priority = 2};
      |                                                    ^
/home/USER/dev/issue-repro/MyDeployment/Top/MyDeploymentTopology.cpp:129:38: error: C++ designated initializers only available with ‘-std=c++20’ or ‘-std=gnu++20’ [-Werror=c++20-extensions]
  129 |     configurationTable.entries[2] = {.depth = 100, .priority = 1};
      |                                      ^
/home/USER/dev/issue-repro/MyDeployment/Top/MyDeploymentTopology.cpp:129:52: error: C++ designated initializers only available with ‘-std=c++20’ or ‘-std=gnu++20’ [-Werror=c++20-extensions]
  129 |     configurationTable.entries[2] = {.depth = 100, .priority = 1};
  ```

## Expected Behavior

The auto-generated code should compile without any failures

LeStarch, 2025-03-12:
@thomas-bc this is a cookie-cutter template update, yes?

thomas-bc, 2025-03-12:
If the solution we choose is not to use this (supposed?) c++20 feature, then yes. But I'd like to understand why this error is not popping up in CI (is it just a WSL-compilers issue?)

LeStarch, 2025-03-12:
I think it is the addition of the pedantic flag.  This will soon become standard.

closed 2025-04-02 by thomas-bc