nasa/fprime #3373: Autogenerated Test Code Uses `FwEnumStoreType` in Some Command Handles

LeStarch, 2025-03-13:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|   |
---
## Problem Description

Found in `Svc/DpManager`.

The autocoded base class uses `FwEnumStoreType` for the following call.  However, I expected it to use `FwOpcodeType` as that is consistent with the command handlers.

```
    this->sendCmd_CLEAR_EVENT_THROTTLE(instance, cmdSeq);
```

bocchino, 2025-03-19:
On further investigation, I'm not sure that there is any problem here. The intent of the `instance` argument is to provide a component instance, not an opcode. The opcode is known from the `sendCmd` function that is called, and it should be provided by the auto-generated code inside `sendCmd`. `FwEnumStoreType` is the correct type of a component instance.

jwest115, 2025-03-24:
@LeStarch just following up on the above, does Rob's comment sound ok or do you expect `FwOpcodeType` to be used instead?

closed 2025-03-24 by LeStarch