nasa/fprime #3419: Cannot use TlmPacketizer because of wrong cookiecut Packets.xml?

rdx-aegse, 2025-03-27:
I fprime-bootstrapped a new project and proceeded to swap TlmChan for TlmPacketizer carefully following all commented instructions in the Top folder of my created deployment; however the compilation fails. When I look through the logs, I can see some interesting lines:

[image]

[image]

[image]

When I compare RefPackets.xml in the Ref deployment of the fprime submodule created by fprime-bootstrap (v3.6.2 apparently, by the way), I can see RefPackets.xml had the prefix "Ref." for every channel name, whereas the {deployment}Packets.xml created by fprime-bootstrap did not have the prefix "{deployment}.". 

I'm going to add them in the hopes it fixes this error because it's my first lead, but it does feel like a bug from the cookiecutter, doesn't it?

rdx-aegse, 2025-03-27:
Confirm it builds fine when prefixing all channels in demoPackets.xml with "demo."

thomas-bc, 2025-03-27:
Looks like an oversight indeed – would you like to submit a PR to fix it?

The template is here: https://github.com/nasa/fprime-tools/blob/devel/src/fprime/cookiecutter_templates/cookiecutter-fprime-deployment/%7B%7Bcookiecutter.deployment_name%7D%7D/Top/%7B%7Bcookiecutter.deployment_name%7D%7DPackets.xml ; you can use `{{cookiecutter.deployment_name}}`

thomas-bc, 2025-03-31:
FWIW, the Packets.xml will soon be deprecated in favor of the new FPP packets spec.

rdx-aegse, 2025-03-31:
Thanks for the warning. I'll probably make a merge request when I find the time to wrap up and publish the application code.
That will use fprime 3.6.2 so we should be fine until we have to upgrade fprime, but for when we do, will we have a fprime-gds loader for the packet spec similar to PacketXmlLoader but with the new fpp way of defining them?

thomas-bc, 2025-03-31:
Yes precisely! That new loader is actually in review right now, if you're interested https://github.com/nasa/fprime-gds/pull/196