nasa/fprime #3428: Pass GTest arguments with `fprime-util check`

thomas-bc, 2025-03-31:
### Discussed in https://github.com/nasa/fprime/discussions/3424

<div type='discussions-op-text'>

<sup>Originally posted by **Darryl-E** March 27, 2025</sup>
Is there anyway to output as xml file when running unit tests for test reports on a pipeline? </div>

## Issue

Converting this discussion to an issue.

It does appear like `--pass-through` is not piped through for `fprime-util check`. We should either add the pass_through, or if not desirable/possible provide a way to pass args to the UT executable. 

Good place to start is looking at that function: https://github.com/nasa/fprime-tools/blob/789610c471ec452e56f54246ecbfe7d606693fee/src/fprime/fbuild/builder.py#L348-L371

thomas-bc, 2025-03-31:
@tylerrussin do you want to take a look at this?

tylerrussin, 2025-03-31:
> [@tylerrussin](https://github.com/tylerrussin) do you want to take a look at this?

Sure, I'll take this one!