nasa/fprime #3441: Add a default set of compiler warnings

bocchino, 2025-04-02:
Now that we have a baseline set of compiler warnings for compiling F Prime, it would be good to enable those by default to each F Prime build. Users who don't want to use particular warnings can turn them off by adding `-Wno-...` to their builds. Doing this would address the issue identified in PR https://github.com/nasa/fprime/pull/3434. In brief the issue is this:

* The current F Prime default build is too permissive for FSW (basic compilation with no warning flags)
* Users can add their own flags, but doing this is error-prone. It's easy to write CMake code that looks like it's adding warnings but isn't; and it's hard to tell when this is happening, because "check warnings and pass" produces the same output as "don't check warnings."

The baseline set of warnings is here:

https://github.com/nasa/fprime/blob/8a4bef243bfdc8342599581b472366dba808fabf/CMakeLists.txt#L24-L35

LeStarch, 2025-05-20:
Make this work on compilers, respect usage requirements in CMake.  Should act as a default set.

LeStarch, 2025-07-23:
See #3924

LeStarch, 2026-09-30:
Done

closed 2026-09-30 by LeStarch