nasa/fprime #3442: fprime-vxworks undefined symbols

kevin-f-ortega, 2025-04-02:
fprime-vxworks has a script https://github.com/fprime-community/fprime-vxworks/blob/devel/cmake/utilities/undefined-symbols.py to detect undefined symbols. This script is broken. It reports missing symbols that are not missing. Example

```bash
rosemurg@nuc rosemurg-test-do-feat-cmd:fsw$ python3 /home/rosemurg/dev/pfsoc-fsw-tester/lib/fprime-pfsoc-vxworks7/fprime-vxworks/cmake/utilities/undefined-symbols.py build-fprime-automatic-vxworks/bin/vxworks/PfSocTester  /srv/tftpboot/vxWorks_icicle_raw_kernel
[ERROR] Missing symbols detected:
    U msgQSend
    U _Assert
    U vmLibInfo
    U __wr_register_frame_info
    U __cxa_atexit
    U shutdown
    U intContext
    U strncat
    U semTake
    U vxbSysClkLibInit
```

Here is a bash script that gives the expected results

## script

```bash
$ cat check-undef-fsw-symbols
#!/usr/bin/env bash

set -euo pipefail

readonly fsw_binary="$1"
readonly defined_in_bsp="$(nm /srv/tftpboot/vxWorks_icicle_raw_kernel | awk '{print $3}' | sort -u)"
readonly undefined_in_fsw="$(nm "$fsw_binary" | grep ' U ' | awk '{print $2}' | sort -u)"

grep -Fxv -f <(echo "$defined_in_bsp") <(echo "$undefined_in_fsw")
```

## results

```bash
$ ./check-undef-fsw-symbols build-fprime-automatic-vxworks/bin/vxworks/PfSocTester
osal_evr_assert_0
_ZN3Gnc11FcEvrHelperC1EPN8PfVceSvc5IpMgrE
_ZN3Gnc20set_gnc_evr_instanceEPNS_5FcEvrE
```

kevin-f-ortega, 2025-04-02:
cc @pcrosemurgy

pcrosemurgy, 2025-04-02:
Thanks @kevin-f-ortega! My recommendation is to abandon `nm` and instead use `wr-ld` to check if the Vx kernel satisfies all unresolved symbols in the FSW binary. This way we're not processing symbol names as text in a script.

pcrosemurgy, 2025-04-02:
`ld` (`wr-ld`) should be designed for this exact use case :)