nasa/fprime #3454: Build failure after updating to latest fprime devel commit

celskeggs, 2025-04-03:
| | |
|:---|:---|
|**_F´ Version_**| 0659f0940b6a570a116e80cb785e715c1543e8b9 |
|**_Affected Component_**| build system |
---
## Problem Description

When I update my fprime version from dcfc109b5bc917115fead652e9ba68cb15900858 to 0659f0940b6a570a116e80cb785e715c1543e8b9 (the latest commit at this moment), I stop being able to build my project due to a "No build root found" error.

```
# fprime-util generate --ninja
[INFO] Generating build directory at: /workspaces/[...]/build-fprime-automatic-[...]
[INFO] Using toolchain file /workspaces/[...]/cmake/toolchain/[...].cmake for platform default
-- The C compiler identification is Clang 19.1.1
-- The CXX compiler identification is Clang 19.1.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - failed
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - failed
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- [python3] python3 found at: /usr/bin/python3
-- [fpp-tools] fpp-depend found at: /usr/local/bin/fpp-depend
-- [fpp-tools] fpp-to-xml found at: /usr/local/bin/fpp-to-xml
-- [fpp-tools] fpp-to-cpp found at: /usr/local/bin/fpp-to-cpp
-- [fpp-tools] fpp-to-dict found at: /usr/local/bin/fpp-to-dict
-- [fpp-tools] fpp-locate-defs found at: /usr/local/bin/fpp-locate-defs
-- [FPRIME] Module locations: /workspaces/[...]/lib/fprime;/workspaces/[...];/workspaces/[...]
-- [FPRIME] Configuration module: /workspaces/[...]/config
-- [FPRIME] Installation directory: /workspaces/[...]/build-artifacts
-- Target build toolchain/platform: [...]/[...]
-- Including /workspaces/[...]/cmake/platform/[...].cmake
-- [LIBRARY] Including library [...] at /workspaces/[...]
-- [sub-build] Performing sub-build: info-cache
-- [sub-build] Generating: info-cache with target/fpp_locs;target/fpp_depend
CMake Error at lib/fprime/cmake/utilities.cmake:595 (message):
  [sub-build] Failed to generate: info-cache:

  CMake Error at lib/fprime/cmake/utilities.cmake:426 (message):

    No build root found for: /workspaces/[...]

  Call Stack (most recent call first):

    lib/fprime/cmake/utilities.cmake:461 (get_nearest_build_root)
    lib/fprime/cmake/FPrime.cmake:212 (get_module_name)
    CMakeLists.txt:29 (fprime_setup_included_code)

  -- The C compiler identification is Clang 19.1.1

  -- The CXX compiler identification is Clang 19.1.1

  -- Detecting C compiler ABI info

  -- Detecting C compiler ABI info - failed

  -- Detecting C compile features

  -- Detecting C compile features - done

  -- Detecting CXX compiler ABI info

  -- Detecting CXX compiler ABI info - failed

  -- Detecting CXX compile features

  -- Detecting CXX compile features - done

  -- [python3] python3 found at: /usr/bin/python3

  -- [fpp-tools] fpp-depend found at: /usr/local/bin/fpp-depend

  -- [fpp-tools] fpp-to-xml found at: /usr/local/bin/fpp-to-xml

  -- [fpp-tools] fpp-to-cpp found at: /usr/local/bin/fpp-to-cpp

  -- [fpp-tools] fpp-to-dict found at: /usr/local/bin/fpp-to-dict

  -- [fpp-tools] fpp-locate-defs found at: /usr/local/bin/fpp-locate-defs

  -- [FPRIME] Module locations:
  /workspaces/[...]/lib/fprime;/workspaces/[...];/workspaces/[...]

  -- [FPRIME] Configuration module: /workspaces/[...]/config

  -- [FPRIME] Installation directory: /workspaces/[...]/build-artifacts

  -- Target build toolchain/platform: [...]/[...]

  -- Including
  /workspaces/[...]/cmake/platform/[...].cmake

  -- [LIBRARY] Including library [...] at
  /workspaces/[...]

  -- [LIBRARY] Adding modules from F´ framework

  -- Configuring incomplete, errors occurred!
Call Stack (most recent call first):
  lib/fprime/cmake/sub-build/sub-build.cmake:40 (execute_process_or_fail)
-- Configuring incomplete, errors occurred!
  lib/fprime/cmake/FPrime.cmake:163 (run_sub_build)
  lib/fprime/cmake/FPrime.cmake:247 (fprime_initialize_build_system)
  CMakeLists.txt:13 (include)

[ERROR] CMake erred with return code 1. Partial build cache remains. Run purge to clean-up.
```

## Context / Environment

Execute `fprime-util version-check` and share the output.

```
[root@c473577b197b lib]# fprime-util version-check
Operating System: Linux
CPU Architecture: x86_64
Platform: Linux-4.18.0-553.34.1.el8_10.x86_64-x86_64-with-glibc2.34
Python version: 3.9.21
CMake version: 3.26.5
Pip version: 21.3.1
Pip packages:
    fprime-tools==3.6.1
    fprime-gds==3.6.1
    fprime-fpp-*==3.0.0a7
Project submodules:
    https://github.com/nasa/fprime.git @ v3.6.0-53-g0659f0940
```

## How to Reproduce

See above.

## Expected Behavior

Updating to the latest version of F Prime should not fail my build. If something significant changed that I now need to accommodate in my project, I expect to find something in the pull request that introduced the problem (#3422) that explains what I need to change.

thomas-bc, 2025-04-07:
This is the reason we discourage users from using `devel`, and stick to released version. Development goes into devel that may break things, and those required changes are called out in each release. 
For what it's worth, the [Update Instructions Needed](https://github.com/nasa/fprime/issues/created_by/e?q=%20label%3A%22Update%20Instructions%20Needed%22%20) label on PRs is our way of saying "This is a breaking change", which was added to https://github.com/nasa/fprime/pull/3422. We may want to rename that so a simpler "Breaking Changes"

What we may want to improve on is that rather than sitting down during a release and writing up all breaking changes; breaking changes should be written up in each PR that introduces them. Come a release, one can simply aggregate them. I know that GitHub also has mechanism to automatically pick-up specific labels and aggregate them under nicely organized sections in GitHub Release notes ([see docs here](http://docs.github.com/en/repositories/releasing-projects-on-github/automatically-generated-release-notes)).

thomas-bc, 2025-04-07:
https://docs.github.com/en/repositories/releasing-projects-on-github/automatically-generated-release-notes#example-configurations this looks pretty neat

LeStarch, 2025-04-07:
@thomas-bc, yes, we should get into the habit of typing up breaking change notes on each PR.

LeStarch, 2025-04-11:
The issue is that F Prime platforms need to be updated after this fix.  The work for this issue is to clean up the error message so that this is clearly the source of the issue.

closed 2025-05-10 by LeStarch