nasa/fprime #3455: Support Array lengths above 256 elements

kubiak-jpl, 2025-04-04:
| | |
|:---|:---|
|**_F´ Version_**| 3.6.2 |
|**_Affected Component_**| fpp |
---
## Feature Description

Array types in fpp are limited to 256 elements. This limitation is too low for some use cases

## Rationale

For example, I'd like to serialize a long list of ADC samples (~4096 samples) into a data product. This is not currently possible because of the 256 element limitation

bocchino, 2025-04-04:
To serialize a long array into a data product you can

* Use an array member of an FPP struct
* Use an array record

We can consider supporting longer arrays with FPP array types. However we would need to change the code generation strategy. Right now there is a constructor that provides one argument for each array member. We could say that longer arrays are allowed, but that they have to be initialized by giving all members a single value.

kubiak-jpl, 2025-04-07:
I didn't realize the code gen was different for array types vs. array members. An array member will work for my use case. I'll close this for now

closed 2025-04-07 by kubiak-jpl