nasa/fprime #3458: Time Context Usage and Comparison

LeStarch, 2025-04-07:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|   |
---
## Problem Description

Put a spec in for what is allow in context and can it be used in comparisons.

thomas-bc, 2025-04-11:
Recommendation: context should not be used in Fw::Time comparisons. Times having the same TimeBase should be comparable regardless of what the context value is set to. These comparisons are the ones performed by framework code (as opposed to project code).

LeStarch, 2025-04-21:
Need to fix FpySequencer too (once this is corrected).

LeStarch, 2025-05-20:
Context: scratch-pad, not used on board. Time comparisons should be allowed to ignore config.

LeStarch, 2025-07-03:
@m-aleem assigning you because you just worked on this code (or are).  This one has two steps:

1. Remove context usage/checks/asserts from comparison function in Fw.Time
2. Update FpySequencer to ignore differences in Fw.Time context.

m-aleem, 2025-07-07:
@thomas-bc @LeStarch For the add and subtract helpers, currently those assert if the time contexts differ, which I think we want to remove, but I want to clarify which time context _should_ be assigned to the return value? For example, should we use the time context of first arg (`a` for add `minuend` for sub) as it currently is?

```C++
    Time Time ::
      add(
        const Time& a,
        const Time& b
      )
    {
#if FW_USE_TIME_BASE
      FW_ASSERT(a.getTimeBase() == b.getTimeBase(), static_cast<FwAssertArgType>(a.getTimeBase()), static_cast<FwAssertArgType>(b.getTimeBase()) );
#endif
#if FW_USE_TIME_CONTEXT
      FW_ASSERT(a.getContext() == b.getContext(), static_cast<FwAssertArgType>(a.getContext()), static_cast<FwAssertArgType>(b.getContext()) );
#endif
...
    Time Time ::
      sub(
        const Time& minuend, //!< Time minuend
        const Time& subtrahend //!< Time subtrahend
    )
    {
#if FW_USE_TIME_BASE
      FW_ASSERT(minuend.getTimeBase() == subtrahend.getTimeBase(), static_cast<FwAssertArgType>(minuend.getTimeBase()), static_cast<FwAssertArgType>(subtrahend.getTimeBase()));
#endif
#if FW_USE_TIME_CONTEXT
      FW_ASSERT(minuend.getContext() == subtrahend.getContext(), static_cast<FwAssertArgType>(minuend.getContext()), static_cast<FwAssertArgType>(subtrahend.getContext()));
#endif
```

LeStarch, 2025-07-07:
@timcanham can you answer the above question?

timcanham, 2025-07-10:
That's a good question. I agree about removing the assert for the time context. How about this?

If they are equal, keep it. If they are not, zero them both. If it's in the docs, users can read that and re-assign it after the operation if it's important.

closed 2025-07-22 by LeStarch