nasa/fprime #3473: Update Running on ARM Linux Tutorial

kevin-f-ortega, 2025-04-10:
There is a command to run this

```bash
scp -r build-artifacts/aarch64-linux/<name-of-deployment> <username>@<device-address>:deployment
```

but there is issue if the deployment already exists from a previous class.

I would suggest using this command

```bash
rsync -a --delete build-artifacts/aarch64-linux/<name-of-deployment>/ pi:~/deployment
```

LeStarch, 2025-05-09:
My one hesitation would be that `rsync` may not come preinstalled where scp does.  Perhaps we can just use a different spin of `scp` https://serverfault.com/questions/538368/make-scp-always-overwrite-or-create-directory

LeStarch, 2025-05-20:
Recommendation: just copy binary with explicit destination,

aborjigin, 2025-07-24:
Would this be something we'd want to clarify on the [ARM Linux tutorial](https://nasa.github.io/fprime/Tutorials/CrossCompilationSetup/ArmLinuxTutorial.html)? 
I've been referring to [this page](https://fprime.jpl.nasa.gov/latest/docs/tutorials/cross-compilation/#cross-compilation-tutorial-compiling-for-arm) recently for cross compilation. 
I think that a potential mitigation for now (especially if there are workshops happening before this is officially resolved) could be to add in a note or a "common troubleshooting" section on the page for users to make sure to delete the `deployment` directory that gets generated after running `scp -r build-artifacts/aarch64-linux/<name-of-deployment> <username>@<device-address>:deployment` with `rm -rf <directory>` 

I also think that Kevin's suggestion would probably work and be successful. `rsync` seems to be [pre-installed in most linux distributions and MacOS](https://linuxize.com/post/how-to-use-rsync-for-local-and-remote-data-transfer-and-synchronization/#:~:text=Installing%20Rsync%20The%20rsync%20utility%20is%20pre%2Dinstalled,install%20it%20using%20your%20distribution%27s%20package%20manager.)

aborjigin, 2025-07-25:
I was able to test this on the [F' Cross Compilation Setup Tutorial](https://fprime.jpl.nasa.gov/latest/docs/tutorials/cross-compilation/) 
Running `rsync -a --delete build-artifacts/aarch64-linux/<name-of-deployment>/ <username>@<device-address>:~/deployment` 
Does the same thing as what is currently documented: `scp -r build-artifacts/aarch64-linux/<name-of-deployment> <username>@<device-address>:deployment` 
without the need to delete the `deployment` folder each time. 
Perhaps it is worth to try on ya'll end and make the updates to the tutorial. Seems like a quick fix and would same some effort. 

Also regarding the worry about `rsync` not being pre-installed, in the above comment I mentioned that it seems to come [pre-installed in most linux distributions and MacOS](https://linuxize.com/post/how-to-use-rsync-for-local-and-remote-data-transfer-and-synchronization/#:~:text=Installing%20Rsync%20The%20rsync%20utility%20is%20pre%2Dinstalled,install%20it%20using%20your%20distribution%27s%20package%20manager.) 
I tested this out on my Ubuntu VM, I did no `sudo apt install rsync` or any install of `rsync` of the sort, so I believe in terms of it not being available, perhaps the issue is not as large as previously thought. 
Also since, assuming users are following the tutorial, there is a copy and paste code snippet that sayd `sudo apt install build-essential git g++ gdb cmake python3 python3-venv python3-pip` 
this could potentially be changed to `sudo apt install build-essential git g++ gdb cmake python3 python3-venv python3-pip rsync` if necessary.

thomas-bc, 2025-07-28:
My reservation is that `scp` is universal - lots of people (including the tutorial) will know how to type in that command without even necessarily needing to copy/paste it. `rsync` may be a little more involved. 
The RPIs don't need the deployment folder, they only need the binary itself. `scp path/to/LedBlinkerBinary <username>@<device-address>:~/LedBlinker` should work perfectly no?

aborjigin, 2025-07-28:
The issue with using `scp` is that when users want to upload a new binary to the device the previous binary doesn't get deleted. The reason using `rsync` was suggested was so that users don't need to worry about deleting the previous binary, which could potentially cause bugs.

`scp` does work, but the user must run a `rm -rf` before every new binary upload to the device to keep the project clean. 
For example, if you have already a binary in a directory `LedBlinker` and you run the `scp` command again, it'll create a new directory within the `LedBlinker` directory called `<name-of-deployment>` 
That directory will have the `bin`, `dict`, `lib` that we expect, but it is harder to find, and it gets the project cluttered. 
AFAIK I don't think `scp` has the capability/flags to delete files/directories. 

I personally have run the `scp` command on a new binary without previously deleting the old binary which led to some confusion as to why the changes weren't happening. Although it's a mistake you can easily correct, it's something that can be forgotten, it'll take extra time to fix, and it's also an extra step. 
We can probably get around this and keep the `scp` command by adding more documentation to the page maybe as a "common trouble-shooting" that just says "If you're uploading a new binary, make sure to run `rm -rf <name-of-deployment-folder>` "  

TLDR: Running `rsync -a --delete build-artifacts/aarch64-linux/<name-of-deployment>/ <username>@<device-address>:~/deployment` would do the same as the `scp` command but it'll delete the previous binary already on the device. 
Regarding the deployment folder, I believe it is just a naming convention so that all of the files that get copied over are in a single directory, and users are easily able to run the deployment on the device afterwards.

thomas-bc, 2025-07-28:
> The issue with using scp is that when users want to upload a new binary to the device the previous binary doesn't get deleted.

This is the case currently because we are using `scp --recursive deploymentFolder/`
If we were to use `scp LedBlinkerBinary` (a single file pushed to a unique location), then the old binary would be replaced by the new one.

aborjigin, 2025-07-28:
Oop yes, @thomas-bc just tested your suggestion of just uploading the binary on my end with a raspberry pi and was successful. 
Sorry it took me a bit to wrap my head around and test your suggestion. 
So the update that should be made to the [ARM Linux Tutorial](https://nasa.github.io/fprime/Tutorials/CrossCompilationSetup/ArmLinuxTutorial.html) and the [F' Cross-Compilation Setup Tutorial](https://fprime.jpl.nasa.gov/latest/docs/tutorials/cross-compilation/) would be to update:

- `scp -r build-artifacts/aarch64-linux/<name-of-deployment> <username>@<device-address>:deployment` to become `scp build-artifacts/aarch64-linux/<name-of-deployment>/bin/<name-of-deployment> <username>@<device-address>:deployment` 
- `deployment/bin/<name-of-deployment> -a 0.0.0.0 -p 50000` to become `./deployment -a 0.0.0.0 -p 50000`

closed 2025-08-06 by thomas-bc