nasa/fprime #3474: Standard way to automatically verify that F Prime code formatting is correct in CI

celskeggs, 2025-04-10:
| | |
|:---|:---|
|**_F´ Version_**| n/a |
|**_Affected Component_**| CI |

---
## Feature Description

It's great that F Prime has defined a standard format for C++ code that can be automatically applied with `fprime-util format`.

For official F Prime repositories, we should enforce that the correct formatting is applied before a PR can be merged. In particular, we should enforce this for F Prime platform repositories (like `fprime-vxworks` and `fprime-vorago`). This is true both for repositories on public GitHub and on internal GitHub Enterprise.

I would like the F Prime team to figure out an official, standard way to do this, so that it isn't applied piecemeal for individual platform repositories.

## Rationale

Consistency, quality, and readability.

celskeggs, 2025-04-10:
@kevin-f-ortega here is the ticket, as requested

thomas-bc, 2025-04-11:
We likely want to develop a reusable workflow in https://github.com/nasa/fprime-actions which checks out a repo, retrieves all source files, and checks formatting

Formatting the codebase has been on the radar for a while (https://github.com/nasa/fprime/issues/1984), it's just one of those items that always gets superseded by higher-priority items. 
I looked into it for a minute, should have been easy if it were not for the fact that `clang-format`, for some obscure reason, has some slight variations in the formatting it outputs across different versions of the tool (even when using the exact same style)... so we need to lock down on a specific version

celskeggs, 2025-04-11:
Remember that not all GitHub Enterprise deployments support GitHub Actions. We may need multiple solutions.

thomas-bc, 2025-05-21:
We could use the PyPi distribution https://pypi.org/project/clang-format/  to get around the clang-format versioning issue

thomas-bc, 2025-06-23:
We've added a `fprime-util format --check` option to do this https://github.com/nasa/fprime-tools/pull/263 

Using it in CI as part of https://github.com/nasa/fprime/pull/3778

Projects can re-use the same workflow definition as part of their CI. Should GitHub Actions not be available, it should be pretty straightforward to adapt to any CI platform. I will close this issue, feel free to re-open if you had something else in mind.