nasa/fprime #3478: Implementation issues in LinuxSpiDriver

celskeggs, 2025-04-11:
| | |
|:---|:---|
|**_F´ Version_**| latest |
|**_Affected Component_**| Drv.LinuxSpiDriver |
---
## Problem Description

I've been reviewing the LinuxSpiDriver implementation so that I can create an equivalent F Prime component for another platform. During this review, I noticed two issues that we should consider fixing.

First, this code has a potential out of bounds memory access violation:

```
        tr.tx_buf = reinterpret_cast<__u64>(writeBuffer.getData());
        tr.rx_buf = reinterpret_cast<__u64>(readBuffer.getData());
        tr.len = writeBuffer.getSize();
```

The `spi_ioc_transfer` struct's `len` field provides the length of both `tx_buf` and `rx_buf`. If the caller passes a `readBuffer` with a shorter length than the `writeBuffer`, the kernel may write past the end of `rx_buf`, potentially corrupting unrelated memory. There should be a `FW_ASSERT` here to ensure that the sizes of the buffers are the same.

Second, the `open` function unnecessarily reads back the settings that it writes, without verifying them:

```
        ret = ioctl(fd, SPI_IOC_WR_MODE, &mode);
        if (ret == -1) {
            this->log_WARNING_HI_SPI_ConfigError(device,select,ret);
            return false;
        }

        ret = ioctl(fd, SPI_IOC_RD_MODE, &mode);
        if (ret == -1) {
            this->log_WARNING_HI_SPI_ConfigError(device,select,ret);
            return false;
        }
```

`SPI_IOC_WR_MODE` updates the setting for `MODE` in the Linux kernel SPI driver. `SPI_IOC_RD_MODE` reads that setting back out. It is not clear why the setting is read back out, because it is not actually verified.

If the purpose of this code is to verify that the kernel accepted the provided value of `mode`, then there needs to be a check on the value of `mode` provided by SPI_IOC_RD_MODE. If not, it is not clear why this second ioctl occurs, and it should either be clarified in a comment or removed.

I don't currently have access to any SPI peripherals for Linux devices, so I'm unable to make any fixes myself. I also have no need for these fixes at this time. But I want to capture these two issues so that they can be fixed in the future. Developers of SPI drivers for other platforms are likely to look at and copy this code, so it would be useful for this code to be fully correct and not misleading.

AlesKus, 2025-04-25:
A possible alternative for the first point is something like `tr.len = min(writeBuffer.getSize(), readBuffer.getSize())`. But, I don't know what is preferable FW_ASSERT or min.

For the second point, 7 years ago the this code snippet [looked](https://github.com/nasa/fprime/blob/5ed2caf80fe515eb96a30732424181a84d89b24d/Drv/LinuxSpiDriver/LinuxSpiDriverComponentImpl.cpp#L120) like this:
```C
        ret = ioctl(fd, SPI_IOC_RD_MODE, &mode);
        if (ret == -1) {
            DEBUG_PRINT("ioctl SPI_IOC_RD_MODE fd %d failed. %d\n",fd,errno);
            this->log_WARNING_HI_SPI_ConfigError(device,select,ret);
            return;
        } else {
            DEBUG_PRINT("SPI fd %d RD mode successfully configured to %d\n",fd,mode);
        }
```
Some time ago the DEBUG_PRINT [has been removed](https://github.com/nasa/fprime/commit/706fadf966ffe9c62e35ad396a64a9a8183a5c73#diff-f26408567a783542946bca72183d884a11c52193b4f3e32afc579ece19c12a78L133). And the code got its current state. Therefore, I think this mode reading should be removed.

celskeggs, 2025-04-25:
Related: #3532

LeStarch, 2025-07-03:
@vincewoo you've been working on drivers, want to to work with SPI now?

vincewoo, 2025-08-26:
@celskeggs Ah! Sorry about that misunderstanding, I see what you mean now. I agree that we should have an ASSERT here to ensure the sizes match. Will address it as part of this ticket as well!