nasa/fprime #3524: Nonexistent dependency in MOD_DEPS does not result in build failure

celskeggs, 2025-04-23:
| | |
|:---|:---|
|**_F´ Version_**| v3.6.0ish |
|**_Affected Component_**| build system |
---
## Problem Description

If I define a deployment to depend on a nonexistent component, I don't get any error:

```
set(MOD_DEPS
        Components_Nonexistent
)
```

Example:

```
[build] ninja: no work to do.
[driver] Build completed: 00:00:00.079
[build] Build finished with exit code 0
```

This is a problem, because I might have _intended_ to depend on a library that now isn't being included. (In all likelihood, this _would_ result in a build error, but it would appear to be unrelated to the original cause.)

## Context / Environment

```
# fprime-util version-check
Operating System: Linux
CPU Architecture: x86_64
Platform: Linux-4.18.0-553.47.1.el8_10.x86_64-x86_64-with-glibc2.34
Python version: 3.9.21
CMake version: 3.26.5
Pip version: 21.3.1
Pip packages:
    fprime-tools==3.6.1
    fprime-gds==3.6.1
    fprime-fpp-*==3.0.0a6
Project submodules:
    https://github.com/nasa/fprime.git @ v3.6.0-52-gdcfc109b5
```

## How to Reproduce

1. Modify MOD_DEPS to reference a nonexistent component.
2. Rebuild.
3. Observe that no error (necessarily) occurs.

## Expected Behavior

Broken references should be caught by the build system and result in an error, so that they can be fixed. Otherwise developers may see a line in MOD_DEPS and not realize that it does not refer to anything.

thomas-bc, 2025-04-23:
@LeStarch was this fixed as part of https://github.com/nasa/fprime/pull/3487 ?

LeStarch, 2025-05-10:
@thomas-bc it was!  Good catch!

closed 2025-05-10 by LeStarch