nasa/fprime #3532: Update the Drv.SpiReadWrite port definition & Drv.LinuxSpiDriver to provide more flexibility

bevinduckett, 2025-04-25:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**| Drv.LinuxSpiDriver |
---
## Feature Description

Update the  `Drv.SpiReadWrite` port definition & Drv.LinuxSpiDriver to 
1. Change `Drv.SpiReadWrite`  so multiple write buffers can be specified 
2.  Update  Drv.LinuxSpiDriver  to allow the write buffer & read buffers to be differently sized 

## Rationale
1. Because some devices that use the SPI interface, for example FRAM chips, require command bytes to be prefixed to the payload for write operations and requiring a single write buffer forces SW to copy the data to be written into another location in order to send the command bytes & write payload together 
2. Because some devices that connect via SPI set the SO signal to high-Z during specific interactions (for example, the SO line for FRAM chips is disabled when the master is performing a write operation). Forcing SW to read the junk data in such cases wastes time and memory.

LeStarch, 2025-05-20:
This seems like it is a project specific driver or possibly a need to add additional component.

celskeggs, 2025-05-21:
Linux provides `writev` and `readv` system calls for a reason: vector-based I/O operations are extremely useful in specific cases where performance or memory use is critical. That's not project-specific, so much as a category of projects that could have this need. It seems like we should be able to negotiate a change to the interface that supports project needs and is appropriate to be merged back into the framework.

Otherwise, we risk a fragmentation of different SPI driver interfaces within the F Prime ecosystem.

closed 2026-09-11 by LeStarch