nasa/fprime #3533: Build cache refreshes twice

thomas-bc, 2025-04-29:
| | |
|:---|:---|
|**_F´ Version_**| `devel` |
|**_Affected Component_**|  CMake |
---
## Problem Description

On `devel`, I've been noticing the build cache will every now and then refresh twice. I believe I've also seen it refresh sometimes when it shouldn't (only .cpp files modified).
This has happened both with `fprime-util` and in an IDE, so likely a CMake issue?

Here are logs from attempting to build `Ref` where build cache is refreshed twice:

<details>
<summary>Click to expand logs</summary>

```bash
fprime/Ref$ fprime-util build -j32

-- [python3] python3 found at: /Users/chammard/Work/fp/fprime/fprime-venv/bin/python3
-- [fpp-tools] fpp-depend found at: /Users/chammard/Work/fp/fprime/fprime-venv/bin/fpp-depend
-- [fpp-tools] fpp-to-xml found at: /Users/chammard/Work/fp/fprime/fprime-venv/bin/fpp-to-xml
-- [fpp-tools] fpp-to-cpp found at: /Users/chammard/Work/fp/fprime/fprime-venv/bin/fpp-to-cpp
-- [fpp-tools] fpp-to-dict found at: /Users/chammard/Work/fp/fprime/fprime-venv/bin/fpp-to-dict
-- [fpp-tools] fpp-locate-defs found at: /Users/chammard/Work/fp/fprime/fprime-venv/bin/fpp-locate-defs
-- [FPRIME] Module locations: /Users/chammard/Work/fp/fprime
-- [FPRIME] Configuration module: /Users/chammard/Work/fp/fprime/config
-- [FPRIME] Installation directory: /Users/chammard/Work/fp/fprime/Ref/build-artifacts
-- Target build toolchain/platform: Darwin/Darwin
-- Including /Users/chammard/Work/fp/fprime/cmake/platform/Darwin.cmake
-- Requiring thread library
-- [sub-build] Performing sub-build: info-cache
-- [sub-build] Generating: info-cache with target/fpp_locs;target/fpp_depend
-- [sub-build] Executing: info-cache with fpp_locs
-- [sub-build] Executing: info-cache with fpp_depend
-- [sub-build] Performing sub-build: info-cache - DONE
-- [INFO] to find 'syft' on PATH, please install to generate software bill-of-materials
-- [LIBRARY] Adding modules from F´ framework
-- Adding Library: cmake_platform_unix
-- Adding Library: config
-- Adding Library: Fw_Buffer
-- Adding Library: Fw_Cmd
-- Adding Library: Fw_Com
-- Adding Library: Fw_Dp
-- Adding Library: Fw_Fpy
-- Adding Library: Fw_Log
-- Adding Library: Fw_Logger
-- Adding Library: Fw_Prm
-- Adding Library: Fw_Time
-- Adding Library: Fw_Tlm
-- Adding Library: Fw_Cfg
-- Adding Library: Fw_Comp
-- Adding Library: Fw_CompQueued
-- Adding Library: Fw_Sm
-- Adding Library: Fw_FilePacket
-- Adding Library: Fw_Obj
-- Adding Library: Fw_Port
-- Adding Library: Fw_Ports_CompletionStatus
-- Adding Library: Fw_Ports_Ready
-- Adding Library: Fw_Ports_Signal
-- Adding Library: Fw_Ports_SuccessCondition
-- Adding Library: Fw_SerializableFile
-- Adding Library: Fw_Test
-- Adding Library: Fw_Types
-- Adding Library: Svc_Cycle
-- Adding Library: Svc_Fatal
-- Adding Library: Svc_Ping
-- Adding Library: Svc_PolyIf
-- Adding Library: Svc_Sched
-- Adding Library: Svc_Seq
-- Adding Library: Svc_WatchDog
[... 271 lines cut ...]
[  5%] Built target config
[  5%] Built target Fw_Cfg
[......]
```

</details>

## Context / Environment

```
Operating System: Darwin
CPU Architecture: arm64
Platform: macOS-15.4.1-arm64-arm-64bit
Python version: 3.12.8
CMake version: 3.24.2
Pip version: 24.3.1
Pip packages:
    fprime-tools==3.6.1
    fprime-gds==3.6.1
    fprime-fpp-*==3.0.0a8
```

## How to Reproduce

1. That, I do not know. I haven't been able to reproduce consistently

LeStarch, 2025-05-14:
I was able to reproduce a rescan at the wrong time, but it went away when I upgraded CMake.

Try installing: `pip install cmake==3.31.6`.  This is the latest CMake before they jumped to 4.0.  It may be an issue with the specific tagged CMake version.

LeStarch, 2025-05-14:
It appears the issue is caused by Ai.xml files being marked as CMAKE_CONFIGURE_DEPENDS even when GENERATED.  This means the configure step is dependent on a build step which is dependent on a configure ... ...

thomas-bc, 2025-05-14:
Closing as it appears to have been fixed by https://github.com/nasa/fprime/pull/3602

Will reopen if it pops up again

closed 2025-05-14 by thomas-bc