nasa/fprime #3537: GDS - log and search by ERT

rlcheng, 2025-04-29:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  |
---
## Feature Description

Currently GDS only logs by SCLK, would be nice to have the ability to also log ERT (earth receive time) and search for cmds, evrs, tlm by that as well.

## Rationale

There are use cases where SCLK and ERT may differ or certain tests need to be conducted at a specific SCLK time but ERT would need to be wall clock.

thomas-bc, 2025-04-30:
Possibly a duplicate of 
https://github.com/nasa/fprime/issues/2650
https://github.com/nasa/fprime/issues/2428

lestarch-autobot, 2026-09-30:
This looks like a possible duplicate of #2428 (ERT logging in the GDS). Suggest closing this issue in favor of #2428.

[Written by Devin](https://nasa-jpl-demo.devinenterprise.com/sessions/83546cf842f144298a0b6a302e0cb6d4)

closed 2026-09-30 by LeStarch