nasa/fprime #3544: ASSERT_FILE_ID missing from delegate builds.

timcanham, 2025-04-30:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|   |
---
## Problem Description

When building the delegate files with asserts set to ASSERT_FILE_ID, CMake does not insert the -`DASSERT_FILE_ID=XXX` into the compiler command line.

## Context / Environment

Execute `fprime-util version-check` and share the output.

```
Operating System: Linux
CPU Architecture: x86_64
Platform: Linux-5.15.133.1-microsoft-standard-WSL2-x86_64-with-glibc2.35
Python version: 3.10.12
CMake version: 4.0.0
Pip version: 22.0.2
Pip packages:
    fprime-tools==3.6.0
    fprime-gds==3.6.0
    fprime-fpp-*==2.3.0
Project submodules:
```

## How to Reproduce

1. Do a build with delegates with ASSERT_FILE_ID for asserts.
2. Observe failure in build

## Expected Behavior

Delegate files should compile with the flag.

LeStarch, 2025-05-01:
@vincewoo didn't you fix this one?

vincewoo, 2025-06-04:
Yes, this should have been fixed by #3418 but let me take a look again.

closed 2025-06-11 by LeStarch