nasa/fprime #3545: Start creating a library of generic data structures

bocchino, 2025-04-30:
F Prime needs a library of generic data structures. A good starting set is in the Data Structures course from the F Prime workshop. Those cover many of the use cases for the core F Prime components.

thomas-bc, 2025-05-01:
See https://github.com/fprime-community/fprime-course-materials/blob/master/Flight_Software_Workshop-October_2024/09_Data_Structures.pdf for course mentioned above

celskeggs, 2025-05-06:
This would definitely be valuable. If we could use an off-the-shelf set of generic data structures, such as the ones in ETL, that would be fantastic. (I see that's mentioned in the slides.) No sense in reinventing the wheel.

bocchino, 2025-05-22:
> If we could use an off-the-shelf set of generic data structures, such as the ones in ETL, that would be fantastic.

I agree that ETL could be a good choice for F Prime projects in many cases. For the core framework, I'm envisioning a simple set of templates that provide basic features and are guaranteed to conform to our coding standards. We want to avoid making the core framework dependent on ETL.

celskeggs, 2025-05-27:
@bocchino Would you mind documenting the rationale for why the framework cannot (or should not) be dependent on ETL? I am not very familiar with ETL, but it sounds like the kind of thing we'd want to use, and it's not obvious to me why it would be preferable for us to reinvent the wheel instead of using it.